esphome/esphome #89: Unstable (MQTT) connection - Sonoff T1 2CH

Boosik, 2018-07-20:
Hi, 
I'm experiencing a extremely unstable (MQTT?) connection on multiple Sonoff T1 2CH. Using Home Assistant MQTT discovery. Both Switches use this same config with only difference being name (bedroom/bathroom) and fan/light names.
I tried to experiment with keepalive interval (tried 5s, 10s, 15s, 30s, 60s) but that made no difference.
This issue doesn't occur instantly but approximately after 3-12 hours after flashing and persists untill reflash (I did not tried restart).
I also tried to remove GPIO12 and GPIO5 sensors that didn't help either.
I also tried to put the switches right next to my router to test if it could be caused by weak wifi signal but the issue persists even then.
I think it is caused by either my config or by bug in esphomeyaml as I was using Sonoff Tasmota before migrating to esphomeyaml and everything was fine.

```
esphomeyaml:
  name: bedroom
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: 'redacted'
  password: 'redacted'

mqtt:
  broker: 'redacted'
  username: 'redacted'
  password: 'redacted'
  port: redacted
  
output:
  - platform: gpio
    pin: 12
    id: light
  - platform: gpio
    pin: 5
    id: fan
  - platform: gpio
    pin: 13
    id: led
    inverted: true

binary_sensor:
  - platform: gpio
    id: light_sensor
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      then:
        - light.toggle:
            id: light_light
  - platform: gpio
    id: fan_sensor
    pin:
      number: GPIO9
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      then:
        - fan.toggle:
            id: fan_fan
  - platform: gpio
    id: output_sensor_1
    pin:
      number: GPIO12
      mode: OUTPUT
    on_press:
      then:
        - light.turn_on:
            id: light_led
    on_release:
      if:
        lambda: 'return !id(output_sensor_2).value;'
      then:
        - light.turn_off:
            id: light_led
  - platform: gpio
    id: output_sensor_2
    pin:
      number: GPIO5
      mode: OUTPUT
    on_press:
      then:
        - light.turn_on:
            id: light_led
    on_release:
      if:
        lambda: 'return !id(output_sensor_1).value;'
      then:
        - light.turn_off:
            id: light_led
  
light:
  - platform: binary
    id: light_light
    name: Ložnice
    output: light
  - platform: binary
    id: light_led
    output: led
    
fan:
  - platform: binary
    id: fan_fan
    name: Ložnice
    output: fan

logger:

ota:
```

xtianpaiva, 2018-07-21:
Have a very similar symptom with all my sonoff flashed with esphomeyaml’s self generated binaries. Previously running Tasmota with no issues.

Really like esphomeyaml, hope this can be solved ASAP.

Boosik, 2018-07-23:
I'd flashed my devices with lowered logger level and it made this issue much less severe - but haven't fixed it completely. I'm still experiencing this issue, but with much less disconnections (down from 10+ disconnections per hour to like 1-2 disconnections per hour).

```
logger:
  level: ERROR
```

ezbe, 2018-07-24:
same with Sonoff Dual R2 and Basic R2
[image]
[image]
Disconnections look very similar.

zmrow, 2018-07-24:
How many MQTT devices are on your network?  What broker are you using?

ezbe, 2018-07-24:
@zmrow +-10 devices using mosquito broker

ezbe, 2018-07-24:
[image]
tassmota works fine

Boosik, 2018-07-24:
I have 6 devices and using Mosquitto.

edif30, 2018-07-25:
I'm having this issue as well.  All my 8266 sensors except 1 is using esphomeyaml.  The only one not having this issue is the one not using esphomeyaml.

fcastroruiz, 2018-07-27:
Similar issue with esp01 board 

[image]

zmrow, 2018-07-27:
This is really interesting.  If I can get some time this weekend, I'll try to dig into the code a bit to see if I can find any indication of a problem there.

It looks like @Boosik has experimented with various keep-alive values.

@fcastroruiz, @edif30 : Have either of you played with keep-alive values?

edif30, 2018-07-28:
@zmrow I have not.  What did you have in mind?

I did just update to the latest esphomeyaml and I've noticed a few times that stability has improved.  Maybe.  I didn't really get a chance to dive in yet.  And I've wiped my HA db a few times while working on things and I lost the history so I will let it run untouched tonight and report back if I go a while without a drop.

theres, 2018-07-28:
Seems like I'm experiencing the same issue, the only difference is that I'm using raw `esphomelib` @ `esp32` (`lolin32`). Tested on 3 different boards. I'm using `Mosquitto` broker.  I don't see any big difference after change of the keep alive - it just restart several times in hour (It's connected from a few seconds up to a few minutes). 

`Mosquitto` log:
```
1532783121: New connection from [IP] on port [MQTT PORT].
1532783121: New client connected from [IP] as geekroom-30aea426b9c0 (x, y, u'z').
1532783221: Client geekroom-30aea426b9c0 has exceeded timeout, disconnecting.
1532783221: Socket error on client geekroom-30aea426b9c0, disconnecting.
1532783250: New connection from [IP] on port [MQTT PORT].
1532783250: New client connected from [IP] as geekroom-30aea426b9c0 (x, y, u'z').
1532783260: Client geekroom-30aea426b9c0 has exceeded timeout, disconnecting.
1532783260: Socket error on client geekroom-30aea426b9c0, disconnecting.
1532783262: New connection from [IP] on port [MQTT PORT].
1532783262: New client connected from [IP] as geekroom-30aea426b9c0 (x, y, u'z').
1532783281: Socket error on client geekroom-30aea426b9c0, disconnecting.
1532783289: New connection from [IP] on port [MQTT PORT].
1532783289: New client connected from [IP] as geekroom-30aea426b9c0 (x, y, u'z').
1532783692: Client geekroom-30aea426b9c0 has exceeded timeout, disconnecting.
1532783692: Socket error on client geekroom-30aea426b9c0, disconnecting.
1532783759: New connection from [IP] on port [MQTT PORT].
1532783759: New client connected from [IP] as geekroom-30aea426b9c0 (x, y, u'z').
1532783767: Client geekroom-30aea426b9c0 has exceeded timeout, disconnecting.
1532783767: Socket error on client geekroom-30aea426b9c0, disconnecting.
1532783776: New connection from [IP] on port [MQTT PORT].
1532783776: New client connected from [IP] as geekroom-30aea426b9c0 (x, y, u'z').
1532783933: Client geekroom-30aea426b9c0 has exceeded timeout, disconnecting.
1532783933: Socket error on client geekroom-30aea426b9c0, disconnecting.
1532783933: New connection from [IP] on port [MQTT PORT].
1532783933: New client connected from [IP] as geekroom-30aea426b9c0 (x, y, u'z').
```

Log from serial monitor (very verbose \w removed not-related stuff):
```
[...]
[W][mqtt.client:071]: MQTT Disconnected: TCP disconnected.
[W][mqtt.client:159]: Lost MQTT Client connection!
[I][mqtt.client:095]: Connecting to MQTT...
[I][mqtt.client:095]: Connecting to MQTT...
[W][mqtt.client:071]: MQTT Disconnected: TCP disconnected.
[I][mqtt.client:095]: Connecting to MQTT...
[W][mqtt.client:071]: MQTT Disconnected: TCP disconnected.
[I][mqtt.client:095]: Connecting to MQTT...
[V][mqtt.client:203]: Publish(topic='geekroom/sensor/wiato_geekroom/state' payload='3.41' retain=1)
[W][mqtt.client:213]: Publish failed!
[W][mqtt.client:071]: MQTT Disconnected: TCP disconnected.
[E][mqtt.client:144]:     Can't connect to MQTT... Restarting...
[I][helpers:213]: Forcing a reboot... Cause: 'mqtt'
[V][wifi:141]: [WiFi event] STA stop (num=3 status=255)
[V][wifi:141]: [WiFi event] AP stop (num=14 status=255)
[V][wifi:141]: [WiFi event] AP stop (num=14 status=255)
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11584
entry 0x40078a60
[...]
```

OttoWinter, 2018-07-28:
Yes that's a known issue and I have spent a lot of time debugging it. From what I saw, esphomelib doesn't itself have any software bugs that could cause this - but the Arduino framework has an unstable TCP/IP stack. 

Some time ago, I looked through a bunch of wireshark logs and there seemed to be two issues:
 a) The WiFi stack in the Arduino core and in the ESP SDKs is quite buggy at times. One issue is that sometimes the WiFi connection is lost, but the appropriate callbacks are never called so the system keeps on thinking the WiFi connection is alive -> therefore any MQTT connection attempts fail.
 b) The TCP/IP stack has a bunch of bugs in both the ESP8266 and ESP32 SDKs - nothing on esphomelib's side, but the MQTT sockets keep on randomly disconnecting.

All of these issues are in the frameworks esphomelib uses and so esphomelib itself unfortunately cannot do anything to solve that. What can help though is experimenting with different arduino framework versions as per https://esphomelib.com/esphomeyaml/guides/faq.html#my-node-keeps-reconnecting-randomly 

The two framework versions that have worked the best for me are:
 * the latest arduino version as in the FAQ
 * On the ESP8266 `espressif8266@1.5.0` - Also the same tasmota uses.

(side note: `keepalive` should only help with reducing the length of outages, not the number of them)

-------

With those steps I still get about 3-6 short outages per day, but from my previous experience with WiFi (I once worked for a company with evaluating different WiFi frequencies in comparison to bluetooth), that number of disconnects is expected though.

One thing to note is that showing the "unavailable" status is a feature :) - I'm pretty sure tasmota also has a few reconnects per day (though not as many as some of you have), but you're just not seeing them because tasmota doesn't show that with the default settings 😉

vincegre, 2018-07-30:
I wast just looking reason why since I use esphomeyaml on my ESP devices I had that ennoying problem (very present on a Sonoff TH10/16 and lot less on a Sonoff Basic) of switches coming back on randomly and without any reasons :-1: 
Looks like there is no solution for now ? out of using alternative firmwares ?

edif30, 2018-07-31:
@OttoWinter IDK and I can't explain it but last week I updated esphomeyaml, refreshed my sensors to add restart switches, and I went from over 10 drops a day down to almost none.

Boosik, 2018-08-06:
@OttoWinter just tried to compile with platform v1.5.0, but it will not compile.
```
PS D:\ESP> esphomeyaml bathroom_sensor.yaml run
[32mINFO [esphomeyaml.config] Reading configuration...[0m
[32mINFO [esphomeyaml.__main__] Generating C++ source...[0m
[32mINFO [esphomeyaml.__main__] Compiling app...[0m
[32mINFO [esphomeyaml.__main__] Running:  platformio run -d bathroom_sensor[0m
Processing bathroom_sensor (platform: https://github.com/platformio/platform-espressif8266.git#v1.5.0; board: nodemcuv2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <Wire> 1.0
|-- <esphomelib> 1.7.0 #dfb57e3
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP Async WebServer> 1.1.1
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <FastLED> 3.1.8 #d416194
|   |   |-- <EspSoftwareSerial>
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Wire> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs\bathroom_sensor\lib061\esphomelib\esphomelib\ota_component.cpp.o
Compiling .pioenvs\bathroom_sensor\lib061\esphomelib\esphomelib\output\ledc_output_component.cpp.o
Compiling .pioenvs\bathroom_sensor\lib061\esphomelib\esphomelib\output\pca9685_output_component.cpp.o
Compiling .pioenvs\bathroom_sensor\lib061\esphomelib\esphomelib\power_supply_component.cpp.o
Compiling .pioenvs\bathroom_sensor\lib061\esphomelib\esphomelib\remote\lg.cpp.o
Compiling .pioenvs\bathroom_sensor\lib061\esphomelib\esphomelib\remote\nec.cpp.o
Compiling .pioenvs\bathroom_sensor\lib061\esphomelib\esphomelib\remote\panasonic.cpp.o
Compiling .pioenvs\bathroom_sensor\lib061\esphomelib\esphomelib\remote\raw.cpp.o
.piolibdeps\esphomelib\src\esphomelib\ota_component.cpp: In member function 'virtual void esphomelib::OTAComponent::setup()':
.piolibdeps\esphomelib\src\esphomelib\ota_component.cpp:42:18: error: 'class ArduinoOTAClass' has no member named 'setPasswordHash'
ArduinoOTA.setPasswordHash(this->password_.c_str());
^
*** [.pioenvs\bathroom_sensor\lib061\esphomelib\esphomelib\ota_component.cpp.o] Error 1
============================================== [ERROR] Took 4.12 seconds ==============================================
```

kenci, 2018-10-04:
maybe this will help: `WiFi.setSleepMode(WIFI_NONE_SLEEP);`

Haven't had any disconnect after i modified wifi_component.cpp and set the sleep mode to none sleep.

OttoWinter, 2018-10-08:
@kenci Wow that seems to be helping a lot. I've put that as the default sleep mode in 1.8.2 and I'm already seeing the disconnects in my network going down. Fingers crossed it will stay that way. Thank you very much! Very much appreciated!

Boosik, 2018-10-11:
I've flashed all my sensors with 1.8.2 and I have not experienced single disconnect in 3 days! This seems to be resolved. I'll give it few more days to be sure.

thundergreen, 2018-10-30:
I'm still facing this issue with latest dev of esphomelib and Arduino Dev branch I played around with keepalive and timeout but won't get any reasonable result.whats the best setting for both ? I'm using esp32 board

OttoWinter, 2018-11-27:
@thundergreen This was using BLE, right? Try with the latest 1.9.2 and 1.9.3 versions, the BLE has gotten a *lot* better there.

I'm closing this issue here as it has gotten a lot better already. For further discussion, I will open a new issue so that all these wifi/mqtt connection issues can be tracked in a central place.

closed 2018-11-27 by OttoWinter