esphome/esphome #90: MQTT availability

ezar, 2018-07-23:
Using manual https://esphomelib.com/esphomeyaml/components/deep_sleep.html I put availability inside mqtt but doesnt works :(

Failed config
  mqtt:
    - Invalid config for [mqtt]: [availability] is an invalid option for [mqtt]. Check: mqtt->availability. (See esp32pool.yaml, line 10).
    - availability: None

zmrow, 2018-07-24:
Could you please provide that section of your config?  You can use triple backticks "```" to enter code.

ezar, 2018-07-24:
```
mqtt:
  broker: '192.168.0.x'
  username: 'user'
  password: 'pass'
  discovery: False
  discovery_retain: False
  availability:
```

OttoWinter, 2018-07-28:
Ah sorry those docs are incorrect (will update them asap). The way to do that would be:

```yaml
mqtt:
  broker: '192.168.0.x'
  username: 'user'
  password: 'pass'
  discovery: False
  birth_message:
  will_message:
```

and the `availability` key is only on a per-component basis:

```yaml
sensor:
  - platform: ...
    # ...
    availability:
```

ezar, 2018-08-07:
yes!