esphome/esphome #92: expected a dictionary for dictionary value @ data['on_boot'][0]['then'][0]['mqtt.publish']

fcastroruiz, 2018-07-27:
Hi:

I want to create a wifi button with mqtt message , with esp01 board, the idea is:

Boot -> connect wifi -> send mqtt message -> wait delay 30 s --> shutdown 

use a button to reset

Something like this: 

https://www.youtube.com/watch?v=ImVK5cGVrpQ&t=36s

when i create on_boot event i receive this error:

INFO [esphomeyaml.config] Reading configuration...
WARNING [esphomeyaml.pins] ESP8266: Pin 6 (6-11) might already be used by the flash interface. Be warned.
Failed config
  esphomeyaml:
    - Invalid config for [esphomeyaml]: expected a dictionary for dictionary value @ data['on_boot'][0]['then'][0]['mqtt.publish']. Got None
extra keys not allowed @ data['on_boot'][0]['then'][0]['payload']. Got None
extra keys not allowed @ data['on_boot'][0]['then'][0]['topic']. Got None. (See esp01.yaml, line 1).
    - binary_sensor: ?
        - platform: status
          name: esp01 estado
      esphomeyaml: [source esp01.yaml:1]
        platform: ESP8266
        board: esp01
        name: esp01
        on_boot: [source esp01.yaml:5]
          priority: -10
          then: ?
            - mqtt.publish: None
              payload: Something happened!
              topic: some/topic
      logger: [source esp01.yaml:26]
        level: DEBUG
      mqtt: [source esp01.yaml:20]
        broker: xxx
        password: xxx
        username: xxx
      ota: [source esp01.yaml:29]
        password: xxx
      sensor: ?
        - platform: adc
          name: esp01 VCC
          pin: VCC
          update_interval: 60s
      switch: ?
        - platform: gpio
          name: esp01 boton
          pin: [source esp01.yaml:44]
            number: GPIO0
        - platform: gpio
          name: esp01 reset
          pin: [source esp01.yaml:48]
            number: 6
      wifi: [source esp01.yaml:16]
        password: xxx
        ssid: xxx

Help me please!

zmrow, 2018-07-27:
@fcastroruiz : Please wrap your config in triple backticks "```" for better readability.  :)

fcastroruiz, 2018-07-27:
Oh !!! Sorry for post noobs code jejeje, now i solved the issue thnks

closed 2018-07-27 by fcastroruiz