esphome/esphome #93: Sonoff as light instate off switch

D3-Bug, 2018-07-29:
Is it possible to configure a Sonoff Basis switch so it will become a light in Home Assistant?

michamichamicha, 2018-07-29:
Yes, you need to define an output with the GPIO pin for the relay and add a binary light component.

D3-Bug, 2018-07-29:
Thanks, going to do that later today, seems simple. 

Perhaps a nice small improvement could be:

switch: <— add option: light
  - platform: gpio
    name: "Sonoff Basic Relay"
    pin: GPIO12

OttoWinter, 2018-07-31:
@D3-Bug I mean theoretically it would be quite easy for me to implement that helper, but I would say I kind of like that `output` needs to be used for two reasons:

 * Some people would expect a `light.gpio` platform to do PWM output (with brightness), even though it would only be a binary light (ON/OFF). So it could be confusing.
 * Also, I kind of like forcing users to use the output component more. Sure, helpers like are nice but this way forces users to learn how to use `id`s.

OttoWinter, 2018-11-27:
Related to https://github.com/OttoWinter/esphomeyaml/issues/243

I do not plan on implementing this feature. I'm therefore closing this.

closed 2018-11-27 by OttoWinter