esphome/esphome #94: Documentation doesnt contain list of commands in a single place

asantaga, 2018-07-29:
Hey 

Firstly nice project! However I noticed, whilst trying to debug my setup, that there isnt a "list of commands to esphomeyaml" documented anywhere.. Looking at your code I determined these

- config  : Display Config
- compile : Compile Code only
- upload : Upload existing firmware.bin to device (useful if your serial port was being used elsewhere (like arduino app)
- logs :  Displays serial output??
- run :  This is documented
-clean : Im assuming this is like a make clean??
-mqtt  : Not sure what this does
-mqtt-fingerprint : Not sure what this does

Can we get some definitive answers here and add it to the getting started guide?

thanks

OttoWinter, 2018-07-31:
... that's the reason why `esphomeyaml --help` exists :)

For example, that currently shows this message:

```bash
usage: esphomeyaml [-h] [-v]
                   configuration
                   {config,compile,upload,logs,run,clean-mqtt,wizard,mqtt-fingerprint,version,dashboard}
                   ...

positional arguments:
  configuration         Your YAML configuration file.
  {config,compile,upload,logs,run,clean-mqtt,wizard,mqtt-fingerprint,version,dashboard}
                        Commands
    config              Validate the configuration and spit it out.
    compile             Read the configuration and compile a program.
    upload              Validate the configuration and upload the latest
                        binary.
    logs                Validate the configuration and show all MQTT logs.
    run                 Validate the configuration, create a binary, upload
                        it, and start MQTT logs.
    clean-mqtt          Helper to clear an MQTT topic from retain messages.
    wizard              A helpful setup wizard that will guide you through
                        setting up esphomeyaml.
    mqtt-fingerprint    Get the SSL fingerprint from a MQTT broker.
    version             Print the esphomeyaml version and exit.
    dashboard           Create a simple webserver for a dashboard.

optional arguments:
  -h, --help            show this help message and exit
  -v, --verbose         Enable verbose esphomeyaml logs.
```

----

The descriptions of each command could of course be improved, but I don't really see a reason why it should be in the docs, as it would just be a mirror of the `esphomeyaml --help` output and yet another thing to maintain with each update.

asantaga, 2018-07-31:
makes sense, perhaps a mention of the --help command, which didnt occur to
me #DUH!

OttoWinter, 2018-11-27:
Fixed now: https://esphomelib.com/esphomeyaml/guides/faq.html#tips-for-using-esphomeyaml

closed 2018-11-27 by OttoWinter