esphome/esphome #95: No idea why this isnt working, esphomeyaml doesnt recognise gpio switches

asantaga, 2018-07-29:
Hey all ,

Whilst I can write/tinker with arduino code I really like this esphomelib/yaml as it will save me loads of time but Im having problems using it properly..

Ive added a reference to GPIO16 to the YAML file, but at runtime it is always marked as "OFF". Ive written some simple arduino code and yes the value is being received correctly (high/low). Any ideas?
My YAML is trivial, my usecase is to echeck the status of my house alarm

esphomeyaml:
name: alarmbox
platform: ESP8266
board: esp07

wifi:
ssid: 'xxxx'
password: 'yyy'

mqtt:
broker: '192.168.0.204'
username: ''
password: ''
switch:

platform: gpio
name: "Alarm Status"
pin:
number: 16
inverted: False
mode: INPUT_PULLDOWN_16
Enable logging
logger:

ota:

Any ideas?

asantaga, 2018-07-29:
not sure if this us useful info.. but I just noticed that once booted the esp8266 doesnt proceed beyond the "first loop finished successfully"

e.g. `0;32m[I][mqtt.client:095]: Connecting to MQTT...[0m
[0;32m[I][mqtt.client:132]: MQTT Connected![0m
[0;35m[C][switch.mqtt:027]: Setting up MQTT switch 'Alarm Status16'[0m
[0;35m[C][switch.mqtt:028]:     Icon: ''[0m
[0;36m[D][mqtt.client:173]: Subscribing to topic='alarmbox/switch/alarm_status16/command' qos=0...[0m
[0;36m[D][switch.mqtt:071]: 'Alarm Status16': Sending state OFF[0m
[0;35m[C][switch.mqtt:027]: Setting up MQTT switch 'Alarm Status15'[0m
[0;35m[C][switch.mqtt:028]:     Icon: ''[0m
[0;36m[D][mqtt.client:173]: Subscribing to topic='alarmbox/switch/alarm_status15/command' qos=0...[0m
[0;36m[D][switch.mqtt:071]: 'Alarm Status15': Sending state OFF[0m
[0;35m[C][switch.mqtt:027]: Setting up MQTT switch 'Alarm Status14'[0m
[0;35m[C][switch.mqtt:028]:     Icon: ''[0m
[0;36m[D][mqtt.client:173]: Subscribing to topic='alarmbox/switch/alarm_status14/command' qos=0...[0m
[0;36m[D][switch.mqtt:071]: 'Alarm Status14': Sending state OFF[0m
[0;35m[C][switch.mqtt:027]: Setting up MQTT switch 'Alarm Status13'[0m
[0;35m[C][switch.mqtt:028]:     Icon: ''[0m
[0;36m[D][mqtt.client:173]: Subscribing to topic='alarmbox/switch/alarm_status13/command' qos=0...[0m
[0;36m[D][switch.mqtt:071]: 'Alarm Status13': Sending state OFF[0m
[0;35m[C][switch.mqtt:027]: Setting up MQTT switch 'Alarm Status5'[0m
[0;35m[C][switch.mqtt:028]:     Icon: ''[0m
[0;36m[D][mqtt.client:173]: Subscribing to topic='alarmbox/switch/alarm_status5/command' qos=0...[0m
[0;36m[D][switch.mqtt:071]: 'Alarm Status5': Sending state OFF[0m
[0;35m[C][switch.mqtt:027]: Setting up MQTT switch 'Alarm Status4'[0m
[0;35m[C][switch.mqtt:028]:     Icon: ''[0m
[0;36m[D][mqtt.client:173]: Subscribing to topic='alarmbox/switch/alarm_status4/command' qos=0...[0m
[0;36m[D][switch.mqtt:071]: 'Alarm Status4': Sending state OFF[0m
[0;33m[W][mqtt.client:213]: Publish failed![0m
[0;35m[C][switch.mqtt:027]: Setting up MQTT switch 'Alarm Status2'[0m
[0;35m[C][switch.mqtt:028]:     Icon: ''[0m
[0;36m[D][mqtt.client:173]: Subscribing to topic='alarmbox/switch/alarm_status2/command' qos=0...[0m
[0;36m[D][switch.mqtt:071]: 'Alarm Status2': Sending state OFF[0m
[0;32m[I][application:090]: Running through first loop()[0m
[0;32m[I][application:106]: First loop finished successfully![0m
[0;32m[I][ota:139]: Boot seems successful, resetting boot loop counter.[0m
`

OttoWinter, 2018-07-31:
So you want to monitor if a pin is pulled HIGH or low? Then you should use `binary_sensor`, not switch. Switches only pull a pin high or low. They do not sense if a pin is pulled high or low externally. So output-only

That's what binary sensors are for: they do sense if a pin is pulled high or low and expose that as a binary sensor in Home Assistant.

```yaml
binary_sensor:
  - platform: gpio
    name: "Alarm Status"
    pin:
      number: 16
      mode: INPUT_PULLDOWN_16
```

----

About your

```bash
e.g. it ends with
�[0;32m[I][application:106]: First loop finished successfully!
�[0;32m[I][ota:139]: Boot seems successful, resetting boot loop counter.
```

That's because in the current configuration your node doesn't do anything except output switch states as per above explanation.

closed 2018-07-31 by OttoWinter

asantaga, 2018-07-31:
thanks ! Makes sense now