esphome/esphome #98: dallas temperature not working on node-mcu and wemos d1 mini

diplix, 2018-08-03:
i set up dallas as described in the docs

```
dallas:
  - pin: GPIO16
```
this did neither work on a node-mcu, nor an wemos d1 mini.

(logging set to verbose)

on the node-mcu i only got 
`[07:53:55][E][sensor.dallas:120]: Requesting conversion failed`

on the d1 it failed more telling (no address output, after that `Requesting conversion failed` again repeatedly):

```
[07:13:50][I][mqtt.client:095]: Connecting to MQTT...
[07:13:51][I][mqtt.client:132]: MQTT Connected!
[07:13:51][V][mqtt.client:203]: Publish(topic='kueche3/status' payload='online' retain=1)
[07:13:51][C][sensor.dallas:042]: Setting up DallasComponent...
[07:13:51][C][sensor.dallas:043]:     Want device count: 0
[07:13:51][D][sensor.dallas:050]: Found sensors:
[07:13:51][C][component:239]:     Update interval: 15000ms
[07:13:51][V][component:058]: set_interval(name='update', interval=15000, offset=5597)
[07:13:51][I][application:090]: Running through first loop()
[07:13:51][I][application:106]: First loop finished successfully!
```

affter installing **espeasy**, the dallas sensor was picked up right away. even if i came back with the dallas sensor address (picked up by espeasy), it was still failing:
```
sensor:
  - platform: dallas
    #address: 0xA40000031F055028
    #28-2a-15-77-91-6-2-72
    address: 0x282A157791060272
    #index: 0
    name: "Temperature #1"
```
```
[07:53:16][I][mqtt.client:095]: Connecting to MQTT...
[07:53:17][I][mqtt.client:132]: MQTT Connected!
[07:53:17][V][mqtt.client:203]: Publish(topic='kueche3/status' payload='online' retain=1)
[07:53:17][C][sensor.dallas:042]: Setting up DallasComponent...
[07:53:17][C][sensor.dallas:043]:     Want device count: 1
[07:53:17][D][sensor.dallas:050]: Found sensors:
[07:53:17][C][sensor.dallas:069]: Device 'Temperature #1':
[07:53:17][C][sensor.dallas:080]:     Address: 0x282A157791060272
[07:53:17][C][sensor.dallas:082]:     Resolution: 12
[07:53:17][C][component:239]:     Update interval: 15000ms
[07:53:17][V][component:058]: set_interval(name='update', interval=15000, offset=7034)
[07:53:17][C][sensor.mqtt:025]: Setting up MQTT Sensor 'Temperature #1'...
[07:53:17][C][sensor.mqtt:027]:     Expire After: 675s
[07:53:17][C][sensor.mqtt:029]:     Unit of Measurement: '°C'
[07:53:17][C][sensor.mqtt:030]:     Accuracy Decimals: 1
[07:53:17][C][sensor.mqtt:031]:     Icon: ''
[07:53:17][C][sensor.mqtt:033]:     Unique ID: 'dallas-282A157791060272'
[07:53:17][V][mqtt.component:065]: 'Temperature #1': Sending discovery...
[07:53:17][V][mqtt.client:203]: Publish(topic='homeassistantconf/sensor/kueche3/temperature_1/config' payload='{"unit_of_measurement":"°C","expire_after":675,"unique_id":"dallas-282A157791060272","name":"Temperature #1","state_topic":"kueche3/sensor/temperature_1/state","availability_topic":"kueche3/status"}' retain=1)
[07:53:17][I][application:090]: Running through first loop()
[07:53:17][E][sensor.dallas:120]: Requesting conversion failed
[07:53:17][I][application:106]: First loop finished successfully!
[07:53:25][E][sensor.dallas:120]: Requesting conversion failed
[07:53:40][E][sensor.dallas:120]: Requesting conversion failed
[07:53:55][E][sensor.dallas:120]: Requesting conversion failed
```

using the same wiring, everything works fine on espeasy. i also tested it with another dallas sensor, same `Requesting conversion failed`.

balk77, 2018-09-16:
Weird, I have mine working on a wemos d1. Did you try another pin? Did you add a 4.7kOhm pull-up as required by the docs?

weiglbua, 2018-12-23:
I encountered the same issue today. Has this issue been resolved or is this still a thing ?

brandond, 2019-01-03:
I have 3 different esp8266 boards running Dallas temperature sensors without any issues. What your whole config look like? Have you tried a different pin? What are you using to supply vcc to the sensors? Do you have a pull-up resistor on the data line?

Unthred, 2019-01-11:
I have the exact same problem so I dug out a different esp8266 board and swapped them. And it works just fine. I now have 5 boards that seem useless..... is there any way I can work out if the pins are labeled wrongly etc?
FYI
Both say DOIT on them
The one that's working has
Model    ESP8266MOD 
Vendor   esp -s
               ISM 2.4Ghz
               PA  +25dBm
               802.11 b/g/n

The ones that don't work have
CE       RoHS
FCC ID: 2AL3B-ESP-F
ISM2.4G   802.11 b/g/n

Other than this both boards look identical

Hope this helps

brandond, 2019-01-11:
If you're using one of the development boards with USB and power circuitry onboard you need to look at the whole thing, not just the ESP module that's mounted to it. The voltage regulators on some boards are super flaky and brown out under the lightest load. The Dallas sensors really need a decent voltage supply to run in one-wire (data+ground) mode, and work much more reliably if you can supply 3.3 or 5v on the vcc pin.