esphome/esphome #102: [Feature Request] Any display support

skeptizzle, 2018-08-16:
Hi Otto.

Thanks for esphomeyaml, it's a treat to work with!

Do you have any timeframe on displays? I'm building a solar heating controller and it would be really awesome to have the temperatures display on the tank instead of having to read them through Home Assistant.

Cheers!

- Jeppe Nielsen

pyrosmiley, 2018-08-21:
Based on this commit 03249780fd84a1b7d56ce08392cb803395f6a872 in the [1.7.0](https://github.com/OttoWinter/esphomeyaml/compare/v1.7.0...master) tag, it would appear that the release is due to be soon! He's been working on it pretty actively and it looks like new releases have usually been around 1-2 months apart. Obviously there's no way to know for sure, but it's definitely not just "to be worked on," it's definitely coming.

skeptizzle, 2018-08-21:
Sounds awesome! Thanks for the heads up.. 😉👍

OttoWinter, 2018-09-19:
Yes, will be next release so closing this issue.

closed 2018-09-19 by OttoWinter