esphome/esphome #107: MAX6675 strange readings

clyra, 2018-08-25:
I followed the example provided and tried to build a temperature sensor using the MAX6675 component and a esp32 board. My yaml looks like this:

esphomeyaml:
  name: forno
  platform: ESP32
  board: nodemcu-32s

[...]

sensor:
  - platform: max6675
    name: "Oven Temperature"
    pin_cs: GPIO25
    pin_clock: GPIO26
    pin_miso: GPIO27
    update_interval: 15s

I'm getting readings of hundreds of degrees for a ambient temperature and a lot of variance. I'm sure this sensor is ok as it gives correct values using a arduino and the example code from adafruit max6675 library. This behaviour was also reported on homeassistant forum (https://community.home-assistant.io/t/esphomelib-a-comprehensive-solution-for-using-esps-with-home-assistant/55349/132).

Arduino reported the temperature around 30C, but from esphomeyaml I got the following:

forno/status online
forno/debug [C][sensor.max6675:042]: Setting up MAX6675Sensor 'Oven Temperature'...
forno/debug [C][esphal:053]:     GPIO Pin 25 with mode OUTPUT
forno/debug [C][esphal:053]:     GPIO Pin 26 with mode OUTPUT
forno/debug [C][esphal:053]:     GPIO Pin 27 with mode INPUT
forno/debug [C][component:239]:     Update interval: 15000ms
forno/debug [C][sensor.mqtt:025]: Setting up MQTT Sensor 'Oven Temperature'...
forno/debug [C][sensor.mqtt:027]:     Expire After: 675s
forno/debug [C][sensor.mqtt:029]:     Unit of Measurement: '°C'
forno/debug [C][sensor.mqtt:030]:     Accuracy Decimals: 1
forno/debug [C][sensor.mqtt:031]:     Icon: ''
forno/debug [I][application:090]: Running through first loop()
forno/debug [I][application:106]: First loop finished successfully!
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=704.0°C
forno/debug [D][sensor.mqtt:091]: 'Oven Temperature': Pushing out value 704.000000 with 1 decimals of accuracy
forno/sensor/oven_temperature/state 704.0
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=1793.2°C
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=576.2°C
forno/debug [W][sensor.max6675:087]: Got invalid value from MAX6675Sensor (0x082F)
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=1986.8°C
forno/debug [W][sensor.max6675:087]: Got invalid value from MAX6675Sensor (0xE065)
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=448.0°C
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=448.0°C
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=448.0°C
forno/debug [I][ota:139]: Boot seems successful, resetting boot loop counter.
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=448.0°C
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=320.0°C
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=320.0°C
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=384.0°C
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=192.0°C
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=384.0°C
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=448.0°C
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=384.0°C
forno/debug [D][sensor.max6675:093]: 'Oven Temperature': Got temperature=384.0°C
forno/debug [D][sensor.mqtt:091]: 'Oven Temperature': Pushing out value 597.616638 with 1 decimals of accuracy
forno/sensor/oven_temperature/state 597.6

clyra, 2018-08-25:
Hi,

I may have found the problem... Changing this line:

val |= uint16_t(this->read_spi_()) << 8;

to:

val <<=8;
val |= uint16_t(this->read_spi_());

in max6675_sensor.cpp seems to fix it. New readings (yes, I heated the tip with a lighter :-) ):

forno/status online
forno/debug [C][sensor.max6675:042]: Setting up MAX6675Sensor 'Oven Temperature'...
forno/debug [C][esphal:053]:     GPIO Pin 25 with mode OUTPUT
forno/debug [C][esphal:053]:     GPIO Pin 26 with mode OUTPUT
forno/debug [C][esphal:053]:     GPIO Pin 27 with mode INPUT
forno/debug [C][component:239]:     Update interval: 15000ms
forno/debug [C][sensor.mqtt:025]: Setting up MQTT Sensor 'Oven Temperature'...
forno/debug [C][sensor.mqtt:027]:     Expire After: 675s
forno/debug [C][sensor.mqtt:029]:     Unit of Measurement: '°C'
forno/debug [C][sensor.mqtt:030]:     Accuracy Decimals: 1
forno/debug [C][sensor.mqtt:031]:     Icon: ''
forno/debug [I][application:090]: Running through first loop()
forno/debug [I][application:106]: First loop finished successfully!
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=25.0°C
forno/debug [D][sensor.mqtt:091]: 'Oven Temperature': Pushing out value 25.000000 with 1 decimals of accuracy
forno/sensor/oven_temperature/state 25.0
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=25.2°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=24.8°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=25.2°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=25.0°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=25.2°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=24.5°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=24.5°C
forno/debug [I][ota:139]: Boot seems successful, resetting boot loop counter.
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=25.2°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=62.0°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=198.5°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=278.8°C
forno/debug [W][sensor.max6675:088]: Got invalid value from MAX6675Sensor (0x205C)
forno/debug [W][sensor.max6675:088]: Got invalid value from MAX6675Sensor (0x205C)
forno/debug [W][sensor.max6675:088]: Got invalid value from MAX6675Sensor (0x2054)
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=638.5°C
forno/debug [W][sensor.max6675:088]: Got invalid value from MAX6675Sensor (0x204C)
forno/debug [W][sensor.max6675:088]: Got invalid value from MAX6675Sensor (0x2064)
forno/debug [W][sensor.max6675:088]: Got invalid value from MAX6675Sensor (0x2054)
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=107.2°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=96.0°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=90.8°C
forno/debug [D][sensor.mqtt:091]: 'Oven Temperature': Pushing out value 111.433334 with 1 decimals of accuracy
forno/sensor/oven_temperature/state 111.4
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=87.2°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=80.2°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=76.0°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=72.0°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=68.5°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=64.8°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=62.5°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=59.8°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=57.2°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=54.8°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=53.0°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=54.2°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=52.5°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=50.8°C
forno/debug [D][sensor.max6675:094]: 'Oven Temperature': Got temperature=49.0°C
forno/debug [D][sensor.mqtt:091]: 'Oven Temperature': Pushing out value 62.833332 with 1 decimals of accuracy
forno/sensor/oven_temperature/state 62.8

OttoWinter, 2018-08-25:
Yes, already fixed in 1.8.0 (along with a move to a new SPI bus system): https://github.com/OttoWinter/esphomelib/blob/master/src/esphomelib/sensor/max6675_sensor.cpp#L55

clyra, 2018-08-25:
> esphomelib/sensor/max6675_sensor.cpp#L55

hi,

Great!  Thank you. Do you have a estimative of when 1.8 will be released?

medicshort87, 2018-08-26:
I changed this and it comes back with
 '[19:09:14][W][sensor.max6675:088]: Got invalid value from MAX6675Sensor (0xFFFF)
'

OttoWinter, 2018-09-19:
@clyra I was away for a while again. Sorry about that. I *want* to release in the next say 3 days or so.