esphome/esphome #108: Devices not connecting to WiFi

aptonline, 2018-08-29:
I've successfully flashed two devices (Wemos D1 Mini and a NodeMCU) but neither devices connect to my wifi network.

Here is my latest attempt on a NodeMCU:

```
esphomeyaml:
  name: bedroom_sensor
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: '******'
  password: '******'

mqtt:
  broker: '192.168.1.12'
  username: '******'
  password: '******'

# Enable logging
logger:

ota:
  password: '******'

binary_sensor:
  - platform: gpio
    pin: D1
    name: "PIR Sensor"
    device_class: motion
    
sensor:
  - platform: dht
    pin: D8
    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Humidity"
    update_interval: 15s
    
web_server:
  port: 80
```

And the output for esphomeyaml dashboard:

```
INFO [esphomeyaml.__main__] Successfully compiled program.
INFO [esphomeyaml.__main__] Uploading binary...
INFO [esphomeyaml.__main__] Running:  esptool.py --before default_reset --after hard_reset --chip esp8266 --port /dev/cu.SLAB_USBtoUART write_flash 0x0 config/bedroom_sensor/.pioenvs/bedroom_sensor/firmware.bin
esptool.py v2.5.0
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:b3:4a:d4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 383808 bytes to 263911...
Writing at 0x00000000... (5 %)
Writing at 0x00004000... (11 %)
Writing at 0x00010000... (29 %)
Writing at 0x0002c000... (70 %)
Writing at 0x00040000... (100 %)
Wrote 383808 bytes (263911 compressed) at 0x00000000 in 23.3 seconds (effective 132.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO [esphomeyaml.__main__] Successfully uploaded program.
INFO [esphomeyaml.__main__] Starting log output from /dev/cu.SLAB_USBtoUART with baud rate 115200
[11:30:00][W][wifi:314]: Timeout while connecting to WiFi.
[11:30:00][I][wifi:227]: WiFi Connecting to 'Portal'...
[11:30:00][C][wifi:244]:     STA Hostname: 'bedroom_sensor'
[11:30:30][E][wifi:094]: Can't connect to WiFi, rebooting...
[11:30:30][I][helpers:213]: Forcing a reboot... Cause: 'wifi'
[11:30:30]
[11:30:30] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[11:30:30]
[11:30:30]load 0x4010f000, len 1384, room 16 
[11:30:30]tail 8
[11:30:30]chksum 0x2d
[11:30:30]csum 0x2d
[11:30:30]v4ceabea9
[11:30:30]~ld
[11:30:30][I][log_component:056]: Log initialized
[11:30:30][C][ota:182]: There have been 8 suspected unsuccessful boot attempts.
[11:30:30][I][application:048]: Running through setup()...
[11:30:30][C][binary_sensor.gpio:019]: Setting up GPIO binary sensor 'PIR Sensor'...
[11:30:30][C][esphal:053]:     GPIO Pin 5 with mode INPUT
[11:30:30][C][wifi:031]: Setting up WiFi...
[11:30:30][I][wifi:227]: WiFi Connecting to 'Portal'...
[11:30:30][C][wifi:244]:     STA Hostname: 'bedroom_sensor'
[11:31:00][W][wifi:314]: Timeout while connecting to WiFi.
[11:31:00][I][wifi:227]: WiFi Connecting to 'Portal'...
[11:31:00][C][wifi:244]:     STA Hostname: 'bedroom_sensor'
[11:31:30][E][wifi:094]: Can't connect to WiFi, rebooting...
[11:31:30][I][helpers:213]: Forcing a reboot... Cause: 'wifi'
```

Any ideas whats going on? previous to using esphomeyaml I used Tasmota and had no issues getting the devices on wifi.

AlexMekkering, 2018-09-02:
I'm experiencing the same issues with a Wemos D1 Mini pro and a NodeMCU v2. I've tried two different Wifi networks and they simply won't connect.
An ESP32 I have connects without problems.

OttoWinter, 2018-09-20:
I know this is a long-standing issue with esphomelib but unfortunately I don't really have a definite fix for it. The issue is somewhere way lower-level than esphomelib operates in the WiFi stack, but one thing that has helped before is [experimenting with the Arduino framework version in use](https://esphomelib.com/esphomeyaml/components/esphomeyaml.html#using-latest-arduino-framework)

aptonline, 2018-09-22:
@OttoWinter I'm happy to report that your suggestion worked for me on a NodeMCU, will try with a D1 Mini and report back.  Many thanks, and welcome back.  Hope your vacation was good.

aptonline, 2018-09-23:
So this also seems to have fixed the issue with D1 Mini connecting to Wi-fi too :) ...

...But the D1 seems to 'reboot' or loose connection after a few mins of being connected.  I have the Wi-Fi status enabled so I can see when its connected and this starts to blink and fing shows its dropped off line after a few mins.

codedmind, 2018-09-24:
@OttoWinter/esphomeyaml
<reply@reply.github.com>
that
workaround work for me too.

For instance can you help me understand how to adapt the example
https://esphomelib.com/esphomeyaml/cookbook/garage-door.html but to work
only with one relay?
press -> open -> press -> stop -> press -> close -> press -> stop etc?

Thanks

à(s) 14:14:

OttoWinter, 2018-09-27:
@codedmind What do you mean by "press -> open -> press -> stop -> press -> close -> press -> stop"? I'm just seeing a bunch of words in some weird configuration :) For example, what should happen when the cover should be opened? What should be done when it is closed? And this is not really the right issue to your question. I'm trying to be more active over on discord now, so it would be great if you could go there.

codedmind, 2018-09-27:
@OttoWinter  where in discord?

My garagem door works by press (like a remote) only have one button to operate.
So
- if the door is open, one press will close the door
- if the door is closed, one press will open the door

OttoWinter, 2018-11-27:
Ok so this has gotten a lot better in recent versions as far as I know.

Closing this issue here as partially resolved. Also creating a new issue so that all these wifi/mqtt connection issues have a central issue to track on.

closed 2018-11-27 by OttoWinter