esphome/esphome #112: Automation conditions do not work with template switch

frog32, 2018-08-31:
[Conditions](https://esphomelib.com/esphomeyaml/guides/automations.html?highlight=condition#automation-conditions) can not be used inside a template switches turn_on_action and turn_off_action since they require a list while conditions are formulated as a dict.

OttoWinter, 2018-09-20:
Starting with the next release, you will be able to do the following:

```yaml
switch
  - platform: template
    name: "Coffee Machine"
    icon: "mdi:coffee"
    lambda: 'return id(power_led).value);'
    turn_on_action:
      - if:
          lambda: 'return id(power_led).value;'
        then:
          # do a
          - switch.turn_on: power_btn
        else:
          # do b
          - switch.turn_off: power_btn
```

closed 2018-09-20 by OttoWinter

frog32, 2018-10-27:
Great but there is some problem with using if actions here:

```
  - platform: template
    name: "Test Switch"
    icon: "mdi:coffee"
    lambda: 'return (id(power_led).value);'
    turn_on_action:
      - if:
          lambda: 'return !(id(power_led).value);'
        then:
          - switch.turn_on:
              id: power_btn
          - delay: 2s
          - switch.turn_off:
              id: power_btn
        else:
    turn_off_action:
      - if:
          lambda: 'return (id(power_led).value);'
        then:
          - switch.turn_on:
              id: power_btn
          - delay: 2s
          - switch.turn_off:
              id: power_btn
```

Gives me the following error when compiling:

```
Failed config
 switch.template: 
   - Invalid config for [switch.template]: Cannot have two actions in one item. Key if overrides then! for dictionary value @ data['turn_off_action']['then']. Got None
Cannot have two actions in one item. Key if overrides then! for dictionary value @ data['turn_on_action']['then']. Got None
extra keys not allowed @ data['turn_off_action'][0]['action_id']. Got None
extra keys not allowed @ data['turn_on_action'][0]['action_id']. Got None. (See /config/esphomeyaml/test_light.yaml, line 126). 
    - platform: template
      icon: mdi:coffee
      lambda: return (id(power_led).value);
      name: Test Switch
      turn_off_action: ?
        - action_id: None
          if: [source /config/esphomeyaml/test_light.yaml:141]
            lambda: return (id(power_led).value);
          then: ?
            - action_id: 
              switch.turn_on: ?
                id: power_btn
            - action_id: 
              delay: 2 s
            - action_id: 
              switch.turn_off: ?
                id: power_btn
      turn_on_action: ?
        - action_id: None
          if: [source /config/esphomeyaml/test_light.yaml:132]
            lambda: return !(id(power_led).value);
          then: ?
            - action_id: 
              switch.turn_on: ?
                id: power_btn
            - action_id: 
              delay: 2 s
            - action_id: 
              switch.turn_off: ?
                id: power_btn
```