esphome/esphome #113: Use a different theme for documentation

danobot, 2018-09-03:
Please use a more appropriate theme for the documentation. It is really difficult to navigate given there is no home/back button or menu structure.
I find i have to modify the URL manually in order to get back to the root page at
https://esphomelib.com/esphomeyaml/

Is there any way you generate the docs using a different Sphinx theme? Surely there are better ones out there, maybe specifically tailored for project documentation?

OttoWinter, 2018-09-20:
Sure, I also hate it. One of the steps I want to do first is get a logo for the project which would enable the docs to have a home button at the top. Theoretically the theme also supports sidebars but I have not yet been able to enable those. A PR would of course be welcome :)

OttoWinter, 2018-09-27:
It's a *bit* better now. There's still a lot of room for improvement but I'm struggling with sphinx a bit.

danobot, 2018-11-03:
Much better, thanks :) [Mkdocs](https://www.mkdocs.org/) is worth checking out.

OttoWinter, 2018-11-27:
esphomelib's documentation codebase is already some 30 thousand lines long, so switching to another documentation system would be very painful. I do not plan to do so (plus I started to learn more about the sphinx extensibility using plugins; a lot of issues can be resolved that way).

closed 2018-11-27 by OttoWinter