esphome/esphome #115: Discussion: standardised MQTT topics for on_click/on_double_click

apeeters, 2018-09-04:
Esphomelib/yaml already provides a standardised set of MQTT topics for sensor definitions and its state changes. In this topic I would like to discuss similar standard topics for the on_click and on_double_click events. Implementing this currently requires manual specification of the topic. It would be nice if esphomelib/yaml included a sensible default.

Documentation: https://esphomelib.com/esphomeyaml/components/mqtt.html#mqtt-publish-action

OttoWinter, 2018-09-25:
I don't really want to make the binary sensor component send click MQTT messages by default for a number of reasons:

* In general, Each MQTT integration is meant to be bare-minimums so that it shows up inside Home Assistant. Everything else should be added manually.
* This would effectively double the network load of binary sensors for every user. The upside is that a few people who need this can save a few lines of YAML. Especially with spotty WiFi connections the increased network load can quickly cause disconnection problems.
* Another question would be what duration parameters this "default" `on_click` MQTT message should take. What happens if a user defines an `on_click` action manually, is the MQTT still sent then?

apeeters, 2018-09-25:
Indeed, sending MQTT messages for these click events should not be enabled by default. It would however be very useful to be able to enable them without having to explicitly specify the topic and/or payload. For example suppose this config could suffice:

```
binary_sensor:
  - platform: gpio
    # ...
    on_click:
      then:
        - mqtt.publish:
```

Esphomeyaml could provide defaults for topic and payload, for the duration (`min_length` and `max_length` parameters) defaults are already provided.

OttoWinter, 2018-09-27:
I mean in theory, yes that would be an option. But unfortunately that will be very difficult to do with esphomeyaml's architecture. 

When the `mqtt.publish` "validator" is called (which would populate the default data, if this feature would work), it has no knowledge of the `on_click` action that initiated the validation. So it would be really difficult and messy to do :(

OttoWinter, 2018-11-27:
Closing this as I think implementing that would quickly become very difficult to maintain and the code would be very messy. I think users can write a few lines. If anything, the way to fix this issue would be to make the syntax for publishing messages more concise and/or allow for a syntax with less repetition.

closed 2018-11-27 by OttoWinter

apeeters, 2019-02-20:
Irrelevant now we have native HA integration.