esphome/esphome #117: Updates

rlust, 2018-09-19:
Otto, this is a fine piece of software and so useful!
Can we expect further updates?

Thanks!

OttoWinter, 2018-09-20:
Yes you can. I just have a lot of other stuff in life to work on (university, work, [ioi](https://ioi2018.jp/)) and those things will always be my priority. My development for esphomelib/yaml is always in bigger batches when I have a week-end/a few days free. So for the future, expect the same kind of development cycle of a big update every few months and then lots of silence :)

apeeters, 2018-09-20:
Suggestion: since the esphomelib/yaml community seems to be quite active, can't we open up development with multiple committers? That will allow for more continuous development and might help you getting trough all suggestions/improvements faster?

What are your thoughts about this?

olskar, 2018-09-25:
I also think that is a good idea @apeeters @OttoWinter , perhaps not necessarily for feature implementations but for bugs like below, it has been fixed since June but I can not still use my BH1750 until a new release is made.

https://github.com/OttoWinter/esphomelib/pull/98

closed 2018-11-27 by OttoWinter