esphome/esphome #124: cannot compile

patraff, 2018-09-25:
I have been belting at this for about 4 weeks and despite numerous beers and swear words for the life of me I can't work out where the issue is.
setup
hassio on ubuntu 16.04 running on intel nuc
esphomeyaml 1.8.0
adafruit feather huzzah flashed with espeasy mega 4096 build 06/06/18 
mqtt server running on same machine as hassio using hassio addon mqtt

Without using esphomeyaml I can connect a hcsr04 to the feather and receive mqtt signals into hassio and display them

If i try to migrate this setup to esphomeyaml using the addon i cannot compile.
i have tried both OTA and the serial port.
All i get is the error along lines of
 Error: [API] Could not connect to PlatformIO API Service. Please try later

Have I missed something simple somewhere?

patraff, 2018-09-26:
Update
thought I would bring back my pi3 b+ and see what happens.
The Pi compiles with no issues.
So the problem is somewhere in the ubuntu 16.04 setup.
I remember having real issues with ubuntu server 18.04 and having to buggerise around with network config files. i'm wondering if an update on 16.04 has given the same issue?

OttoWinter, 2018-09-26:
I think someone had similar problems some time ago, but I can't find it anymore. For them I think it had to do with the DNS settings on the machine. For some reason DNS resolves weren't working in the docker container.

Sorry, I can't really help with that. I think it's more of an issue with your setup and not really esphomelib. Also, as I can't replicate the error I also can't really help :(

TheJulianJES, 2018-09-27:
I also had the same issue but I think that it could be fixed by the help in: https://github.com/home-assistant/hassio/issues/399

OttoWinter, 2018-11-13:
Yes it seems to be home-assistant/hassio#399. Closing this here.

closed 2018-11-13 by OttoWinter