esphome/esphome #127: ESP32 BLE Tracker issues

onkytonk, 2018-09-27:
Hi!
I recently update to 1.8.0 which address the partition size issues to run the BLE tracker on an ESP32. 
Everything compiles and uploads fine to the ESP32 module, but then I don't see it scanning or discovering any devices. 

If I use Arduino IDE, and use the BLE tracker example sketch, my ibeacons (among other devices) are discovered. So I know the problem isn't the ESP32 module, or the ibeacons I'm using. 

Even if I add the ibeacon MAC address, it appears in Home assistant, but only ever shows as off. 

I am using Windows 10.
I have tried on both a Node-MCU and a devboard ESP32.

OttoWinter, 2018-09-27:
Hmm, I need more information to debug this. Could you please post the arduino sketch file of your working code and post the USB log output that esphomelib shows?

onkytonk, 2018-09-29:
Hey Otto,

Here is the serial output:
```

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
[1B][0;32m[I][log_component:055]: Log initialized[1B][0m
[1B][0;35m[C][ota:186]: There have been 1 suspected unsuccessful boot attempts.[1B][0m
[1B][0;32m[I][application:051]: Running through setup()...[1B][0m
[1B][0;35m[C][wifi:031]: Setting up WiFi...[1B][0m
[1B][0;32m[I][wifi:239]: WiFi Connecting to 'Home 2.4GHz'...[1B][0m
[1B][0;35m[C][wifi:256]:     STA Hostname: 'espbeacon'[1B][0m
[1B][0;32m[I][wifi:290]: WiFi connected![1B][0m
[1B][0;35m[C][wifi:292]:     SSID: Home 2.4GHz[1B][0m
[1B][0;35m[C][wifi:294]:     BSSID: E4:F4:C6:07:74:85[1B][0m
[1B][0;35m[C][wifi:295]:     Channel: 13[1B][0m
[1B][0;35m[C][wifi:296]:     IP Address: 192.168.1.34[1B][0m
[1B][0;35m[C][wifi:297]:     Subnet: 255.255.255.0[1B][0m
[1B][0;35m[C][wifi:298]:     Gateway: 192.168.1.1[1B][0m
[1B][0;35m[C][wifi:299]:     DNS1: 192.168.1.1[1B][0m
[1B][0;35m[C][wifi:300]:     DNS2: 0.0.0.0[1B][0m
[1B][0;35m[C][ota:027]: Setting up OTA...[1B][0m
[1B][0;35m[C][ota:028]:     port: 3232[1B][0m
[1B][0;35m[C][ota:033]:     hostname: 'espbeacon'[1B][0m
[1B][0;35m[C][mqtt.client:024]: Setting up MQTT...[1B][0m
[1B][0;35m[C][mqtt.client:025]:     Server Address: 192.168.1.13:1883[1B][0m
[1B][0;35m[C][mqtt.client:026]:     Username: 'XXXXXXX'[1B][0m
[1B][0;35m[C][mqtt.client:027]:     Password: 'XXXXXXX'[1B][0m
[1B][0;35m[C][mqtt.client:030]:     Client ID: 'espbeacon-30aea41698a0'[1B][0m
[1B][0;35m[C][mqtt.client:032]:     Discovery prefix: 'homeassistant'[1B][0m
[1B][0;35m[C][mqtt.client:033]:     Discovery retain: true[1B][0m
[1B][0;32m[I][mqtt.client:099]: Connecting to MQTT...[1B][0m
[1B][0;32m[I][mqtt.client:135]: MQTT Connected![1B][0m
[1B][0;35m[C][esp32_ble_beacon:050]: Setting up ESP32 BLE beacon...[1B][0m
[1B][0;32m[I][application:093]: Running thr_controller_init fail
[1B][1;32m[I][application:109]: First loop finished successfully![1B][0m[00]9[1B][0m[1B][0;32m[I][application:109]: First loop finished successfully![1B][0m

[1B][0;33m[W][mqtt.client:071]: MQTT Disconnected: TCP disconnected.[1B][0m
[1B][0;33m[W][mqtt.client:156]: Lost MQTT Client connection![1B][0m
[1B][0;32m[I][mqtt.client:099]: Connecting to MQTT...[1B][0m
[1B][0;32m[I][mqtt.client:135]: MQTT Connected![1B][0m
```

If I do a reboot, I get this the second time:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
[1B][0;32m[I][log_component:055]: Log initialized[1B][0m
[1B][0;35m[C][ota:186]: There have been 2 suspected unsuccessful boot attempts.[1B][0m
[1B][0;32m[I][application:051]: Running through setup()...[1B][0m
[1B][0;35m[C][wifi:031]: Setting up WiFi...[1B][0m
[1B][0;32m[I][wifi:239]: WiFi Connecting to 'Home 2.4GHz'...[1B][0m
[1B][0;35m[C][wifi:256]:     STA Hostname: 'espbeacon'[1B][0m
[1B][0;32m[I][wifi:290]: WiFi connected![1B][0m
[... 42 lines cut ...]
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  BLEScanResults foundDevices = pBLEScan->start(scanTime);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

OttoWinter, 2018-10-04:
Could you try if the fix mentioned here works for you? https://github.com/OttoWinter/esphomeyaml/issues/78#issuecomment-425746566

animavitis, 2018-10-05:
For me this fix https://github.com/OttoWinter/esphomeyaml/issues/78#issuecomment-425746566 is working, but on espressif32@1.0.2 component I2C is not working :/

onkytonk, 2018-10-06:
Success! Seems to be working with espressif32@1.0.2. 

Not sure what the issue is, but I have so far tested it on a NodeMCU ESP32 board. I also have a generic dev board that wasn't working either that I'm happy to test too if need be.

OttoWinter, 2018-10-08:
@onkytonk Ok, that's good to hear! For the BLE always being on. I recently changed something in the code there, and just checked if the discovering worked. Apparently I had a bug that would not send the OFF when the scan was done and the device was not discovered. See https://github.com/OttoWinter/esphomelib/pull/185. Thanks for reporting that!

onkytonk, 2018-10-08:
hey @OttoWinter 
So still having a few issues. 
Firstly, no device names ever show up during the scan.
Secondly, my scan never actually "completes". 
So after the initial on or off value, the value will not change after that, I think because the device is still trying to scan for other devices. 
I've waited over 10 minutes for the scan to "complete" and for my detected ibeacons to turn off, after initially being turned on.

ladefoged81, 2018-10-08:
I've just started using the ble tracker today but seem to experience the same issue as @onkytonk 
I've set it up to track 2 cellphones via bluetooth but they stay "off" eventhough it works with my Home Assistant bluetooth device tracker.

guardmedia, 2018-10-13:
I'm not sure why, but the following change seems to resolve the original issue reported using the latest IDF framework. I'm no longer getting `esp_bt_controller_init failed: 259` if I modify `esp32_ble_tracker.cpp` to use the `btStart()` function instead of manually initializing and enabling the controller. It seems to change the way the linker works by calling this function.

**esp32_ble_tracker.cpp**
```diff
diff --git a/src/esphomelib/esp32_ble_tracker.cpp b/src/esphomelib/esp32_ble_tracker.cpp
index 45c3ad6..46565d7 100644
--- a/src/esphomelib/esp32_ble_tracker.cpp
+++ b/src/esphomelib/esp32_ble_tracker.cpp
@@ -96,6 +96,8 @@ void ESP32BLETracker::ble_core_task(void *params) {
 }

 void ESP32BLETracker::ble_setup() {
+  ESP_LOGCONFIG(TAG, "Setting up ESP32 BLE Tracker...");
+
   semaphore_scan_end = xSemaphoreCreateMutex();

   // Initialize non-volatile storage for the bluetooth controller
@@ -106,19 +108,12 @@ void ESP32BLETracker::ble_setup() {
   }

   // Initialize the bluetooth controller with the default configuration
-  esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
-  err = esp_bt_controller_init(&bt_cfg);
-  if (err != ESP_OK) {
-    ESP_LOGE(TAG, "esp_bt_controller_init failed: %d", err);
-    return;
-  }
-
-  err = esp_bt_controller_enable(ESP_BT_MODE_BLE);
-  if (err != ESP_OK) {
-    ESP_LOGE(TAG, "esp_bt_controller_enable failed: %d", err);
-    return;
-  }
+  if (!btStarted()) {
+    if(!btStart()) {
+        ESP_LOGE(TAG, "btStart failed!");
+        return;
+    }
+  }
```

Tested with the following YAML configuration.
```yaml
esphomeyaml:
  name: crazylongname
  platform: ESP32
  board: mhetesp32devkit
  esphomelib_version: dev
  arduino_version: dev

wifi:
  ssid: 'obnoxiousssid'
  password: 'totallyobviouspassword'
  id: my_wifi

esp32_ble_tracker:
  scan_interval: 300s

binary_sensor:
  - platform: esp32_ble_tracker
    mac_address: AA:BB:CC:DD:EE:FF
    name: "bestnameever"
```

Esiravegna, 2018-11-04:
Using the  espressif32@1.0.2 branch causes the device to reboot constantly on mqtt connection.

OttoWinter, 2018-11-08:
@Esiravegna Please try with the latest beta: https://beta.esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-update-to-the-latest-beta-release - that issue has been fixed there. Plus then you don't need to have `espressif32@1.0.2` in your config anymore.

OttoWinter, 2018-11-08:
Closing this issue as fixed in beta.

closed 2018-11-08 by OttoWinter

Esiravegna, 2018-11-08:
Excellent, I will! Thanks!

ladefoged81, 2018-11-08:
I get these error when compiling with rc or dev docker image. Compiles without errors on 1.8.2
```
.piolibdeps/esphomelib/src/esphomelib/esp32_ble_tracker.cpp:582:53: error: 'len' was not declared in this scope

*** [.pioenvs/esp_test/lib8e5/esphomelib/esphomelib/esp32_ble_tracker.cpp.o] Error 1

```