esphome/esphome #129: Multiple issues with I2C with ESP32

mssaleh, 2018-09-27:
I'm trying to use the I2C bus on the ESP32 with a single sensor and/or multiple sensors/devices.

Here is my setup:
- platform: ESP32 / board: esp32doit-devkit-v1 / arduino_version: latest
- i2c: sda: 21 / scl: 22
- devices used: 
-- BME280 Temperature/Pressure/Humidity sensor
-- BH1750 Illuminance sensor
-- 0.96in OLED display using ssd1306_i2c element

Config:
- exactly as per your documentation, compiled on Windows 10 PC (since I'm using ESP32)
- I tried different wiring schemes with and without pull/up resistors on SDL/SCL lanes.

Issues:
- ssd1306_i2c element never worked! it's only working when I flash it with my own sketch on Arduino IDE. I tried everything: used the display alone without any sensors, with and without "external_vcc", scanning for the I2C address (it surely is 0x3C), tried different fonts and sizes, with shapes or text. It never switches on and never display anything. When I switch to a project with Arduino Nano, the display works perfectly.
- BME280 and BH1750 sensors basically work both individually and in parallel with the address 0x76 and 0x23 respectively. However, they only report the reading via MQTT once at first boot, and then never send another message. I tried with different "update_interval" values (and also without it altogether).

Any suggestions?

OttoWinter, 2018-09-27:
The SSD1306 issue probably the same as https://github.com/OttoWinter/esphomeyaml/issues/128

As for the BME280/BH1750: Have you clicked the text under "See Default Filter." on those pages? - by default the last 15 or so values are averaged before sending them out. So if you have an `update_interval` of 15s, it will still take a few minutes until the next value is sent out via MQTT.

mssaleh, 2018-09-27:
First I apologize for not starting with a proper appreciation for the great effort you put in this project. We often take people like you for granted and we are really spoiled by the wealth of freely-available code on GitHub and other open-source treasures.

Back to the issues: I replaced the ESP32 with a NodeMCU v2 (ESP2866), and the I2C bus worked properly (using pull-up resistors on both SDA and SCL lanes). Now I can get frequent updates in Home Assistant from BME280/BH1750 sensors. However, the SSD1306 display still doesn't work. The ESP2866 discovers it on 0x3C address and the log doesn't show any errors or warnings, but the display stays blank!

I read your reply to issue #128 and if I can be of any help (for testing) please let me know.

OttoWinter, 2018-09-27:
Sure, no problem (if I was too grumpy in my previous response, sorry about that. when maintaining a project like this you learn to be very brief).

It's weird that it wouldn't work with the ESP32. I know the i2c implementation by the arduino framework guys is not very stable on the ESP32, but at least it should show errors if something is not working.

So for the SSD1306, I think I identified *an* issue. Could you test with your SSD1306 if it works with the fix (see #128 for instructions)? Thanks!

OttoWinter, 2018-11-13:
I think this is fixed now (#128)

closed 2018-11-13 by OttoWinter