esphome/esphome #147: PN532 - Only 40% of tags are read

magnusoverli, 2018-10-05:
Hi!

I used to have my PN532 hooked up ESPeasy on I2C. When I read a tag I got a number that varied from 8 to 10 number of digits:
`362449582`, `1432822573` and `99360557` are all real-world examples. 
Every tag I have worked.
I am now using the same PN532 sensor on SPI with ESPhomeyaml. Now, the same cards have different uid´s. On top of that, only about 40% of my cards works. I even bought a new batch of cards, in case some of the old ones had stopped working (my kids are the primary users of this equipment), but it exactly the same.

I´d be interested in knowing where the difference in uid´s comes from, and also why only some of the cards work...

If I need to provide more data on this, please let me know! :-)

OttoWinter, 2018-10-05:
Weird, I cannot find any obvious differences between esphomelib's and espeasy's pn532 code.

But first: The difference in the number comes from the was UIDs work with the RFID protocol. Each tag has it's own semi-unique UID `uid` which has a length of `l` bytes.

When esphomelib receives this uid, it converts each byte to a hexadecimal number and separates the bytes using a hyphen. When espeasy receives a uid, it converts the whole uid to a decimal representation like `1432822573`. So if you convert the hexadecimal number `0x5567232D` to decimal, you will get `1432822573` :)

The reason for this is that the uids `00-56-42-DE` and `56-42-DE` *are* different (because of the data length) but espeasy would treat them the same. This wouldn't happen often but the former is more safe :)

So about the 40% bug: That's not good :( I only have a keyring-tag, a mifare card and a bunch of stick-on NFC tags, all of which work. Looking at the code differences the only thing that may be a bit different is the update frequency - could you try with different `update_interval`s? https://esphomelib.com/esphomeyaml/components/pn532.html (The communication protocol *is* different, SPI vs i2c, but the PN532 has a protocol that is the same across all the supported busses). Also, the reason esphomelib chose SPI is because I've found it to be way more stable. I2C just randomly breaks on ESP32s.

Also, I would like to know more about the symptoms. Is it *only* to those 40% of tags, or do the other ones also have occasional problems? Do these 40% *never* work or do they just work seldomly. And lastly: could you post a few of the ESPEasy ids?

magnusoverli, 2018-10-05:
Thanks for the thorough answer. Appreciated!!

From my experience the last week or so, it is always the same tags that aren't working. I can move the tag close the sensor and hold it for 5 seconds, and a tag that works will be read 2 or 3 times, with an `update_interval` of 2 secs. Tags that aren't working will never be read. The `update_interval` doesn't seem to affect the ability to read the tags that aren't working.

Here is a list of known uid´s that works with ESPhomeyaml, in decimal form:
```
1432822573
1427038509
359142189
2796769313
1427516973
```
Here´s a list of the ones that is never read:
```
362449582
848604196
1668806951
851698468
```

OttoWinter, 2018-10-08:
Thanks! I've stared at the code a couple of times more now and I can't find any issue :(

So my next question would be: Is it only a certain *type* of tag or does it affect them randomly? And do you have a link to those tags so that I can buy&test them myself?

closed 2018-10-13 by OttoWinter