esphome/esphome #148: [Question] Is it possible to Transmit 433Mhz code recieved from MQTT command ?

animavitis, 2018-10-07:
Hi

Is it possible to send MQTT command to esphomeyaml device with code that should be then send via Remote Transmitter Component?? ie.:

From my Home assistant i want to send something like that:
```
        topic: "esphomeyaml/rftransmit/command"
        payload: "{"rc_switch_raw":{"code":"001010011001111101011011","protocol":"1"}}"
```

then i want esphomeyaml to send automatically this as RF code  (without define this code as Switch in eshomeyaml config file).

Any idea how to do this??

OttoWinter, 2018-10-08:
No, that's currently not possible. The reason is mostly because the logic that "converts" the string representation of the code (`001010011001111101011011`) into the actual stuff sent happens in esphomeyaml on your PC. Having the same implementation a second time in esphomelib in another programming language would be a maintaining nightmare.

Maybe in the future, but until that can happen esphomelib would need to get a whole event bus to communicate between components which will be ~~fun~~ to implement :)

rradar, 2018-10-23:
Give a look to OpenMQTTGateway (OMG!) for now :smile: 

https://github.com/1technophile/OpenMQTTGateway

Probably exactly what are you looking for (also works for infrared and in both directions)!