esphome/esphome #153: Compile Error

JAAlperin, 2018-10-08:
Hello Otto.  Thanks for the wonderful esphomeyaml.

I had six nodemcus running fine with code compiled/uploaded by 1.7.0.  One device started to give spurious readings so I decided to load it again.  Upgraded to 1.8.0 then 1 then 2.  Revised yaml, which validates.  But, get this error in the link stage:
collect2: error: ld returned 1 exit status
*** [.pioenvs/nodemcu01/firmware.elf] Error 1 due to many multiple definitions.

I've attached the yaml and the compile log.

I run hass 0.79.3 on hass.io 133 in Docker CE on Raspbian GNU/Linux 9 (stretch) on a B3+

Thank you for any help you can give.

[esphomeyaml yaml validation.txt](https://github.com/OttoWinter/esphomeyaml/files/2455878/esphomeyaml.yaml.validation.txt)
[esphomeyaml compile log.txt](https://github.com/OttoWinter/esphomeyaml/files/2455879/esphomeyaml.compile.log.txt)

JAAlperin, 2018-10-09:
nodemcu01.yaml has esphomeyaml name: nodemcu01.  Duplicate that file to nodemcu01a.yaml with esphomeyaml name: nodemcu01a.

Compile nodemcu01.yaml collects 37 libraries and fails with duplicate definitions.
Compile nodemcu01a.yaml collects 32 libraries and succeeds.

Now what?

brandond, 2018-10-09:
`rm -rf nodemcu01/.pioenvs` and compile again?

JAAlperin, 2018-10-09:
Thank you brandond, but that did not work.

OttoWinter, 2018-10-09:
Puh... I don't really know what this could be. Probably platformio has some old files lying around somewhere. So I guess `rm -rf nodemcu01/.pio*` should hopefully work, or else `rm -rf nodemcu01/` should work (If you've modified your `main.cpp`, you should back that up).

JAAlperin, 2018-10-09:
Deleting the nodemcu01 directory solves the problem.  Library manager rebuilds and only 32 libraries are found.  Thank you, Otto.

I suggest that when the addon is updated that the directories inside esphomeyaml get renamed (to save them and avoid this error).

Great facility, great help, and great to have it working again.

closed 2018-10-09 by JAAlperin