esphome/esphome #154: Display compiles to ".piolibdeps/esphomelib/src/esphomelib/application.h:229:3: error: 'Font' in namespace 'esphomelib::display' does not name a type"

dennis1987, 2018-10-08:
Hi there,

when I compile a yaml with "display" inside, I always run into 

```
Compiling .pioenvs/displaytest/libe37/ESP8266WiFi/WiFiServer.cpp.o
In file included from src/main.cpp:2:0:
.piolibdeps/esphomelib/src/esphomelib/application.h:229:3: error: 'Font' in namespace 'esphomelib::display' does not name a type
display::Font *make_font(std::vector<display::Glyph> &&glyphs, int baseline, int bottom);
^
.piolibdeps/esphomelib/src/esphomelib/application.h:231:3: error: 'Image' in namespace 'esphomelib::display' does not name a type
display::Image *make_image(const uint8_t *data_start, int width, int height);
^
Compiling .pioenvs/displaytest/libe37/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
*** [.pioenvs/displaytest/src/main.cpp.o] Error 1
```
My yaml is:

````
esphomeyaml:
  name: displaytest
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 'ssid'
  password: 'password'

mqtt:
  broker: '192.168.1.1'
  username: ''
  password: ''

# Enable logging
logger:

ota:

web_server:
  port: 80

display:
  - platform: lcd_gpio
    dimensions: 16x2
    data_pins:
      - D7
      - D6
      - D5
      - D4
      - D3
      - D2
      - D1
      - D0
    enable_pin: A0
    rs_pin: D8
    lambda: |-
      it.print("Hello World!");

````

I also tried the ` esphomelib_version: dev` option, but there is the same result. Is there some code for display missing?

I installed esphomeyaml 1.8.2 on macOS High Sierra via pip in a virtualenv using python 2.7.10.

Thanks in advance 
... and a very great job at this project

OttoWinter, 2018-10-08:
https://github.com/OttoWinter/esphomelib/pull/186 - What a single missing line can do :)

could you try again with `esphomelib_version: dev`?

closed 2018-10-08 by OttoWinter

Hilarious83, 2018-10-08:
Hi, I'm having the same problem within the Hass.io addon. Can you tell me how i can fix this? Or is there no fix yet?

Thank you very much

OttoWinter, 2018-10-08:
@Hilarious83 Yes, just put 

```yaml
esphomeyaml:
  # ...
  esphomelib_version: dev
```

into the `esphomeyaml:` part of your configuration.

roadfox, 2018-11-04:
Had the same issue with this conf:
`display:
  - platform: lcd_pcf8574
    dimensions: 20x4
    address: 0x27
    lambda: |-
      it.print("Hello World!");`

chaning to `esphomelib_version: dev` soves the bug and display is working

thanks you so mutch