esphome/esphome #157: Inverting a light?

pschmitt, 2018-10-10:
Hi and thanks for this truly great project. I'm currently migrating all my Tasmota devices to esphomeyaml, the last remaining issue I have that there is no way to "invert" (ie. `ìnverted: true`) a light component.

Here's the relevant config:

```yaml
output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin:
      number: GPIO13
      inverted: false  # turn on by default on boot

light:
  - platform: monochromatic
    id: led
    name: "Sonoff p0: Green LED"
    output: basic_green_led
    default_transition_length: 0s
    # inverted: true  # FIXME This is invalid.
```

... which works fine for having the LED turn on automatically on boot, but sadly the control shown in Home Assistant in inverted.

OttoWinter, 2018-10-10:
I think for that use-case the following would be better and should be preferred:

```yaml
esphomeyaml:
  # ...
  on_boot:
    priority: 50
    then:
      - light.turn_on: led

output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin: GPIO13

light:
  - platform: monochromatic
    id: led
    name: "Sonoff p0: Green LED"
    output: basic_green_led
    default_transition_length: 0s
```

pschmitt, 2018-10-10:
Thanks for your help. I experimented with this type of config and encountered something odd.

For some reason for my Sonoff to behave[*] I need to call the `light.turn_off` service on boot, but later on ie. when having to respond to button presses in order to turn on the LED I need to call `light.turn_on`.

My config is at https://github.com/pschmitt/esphomeyaml-config/blob/master/config/sonoff_p0.yaml
Relevant lines:
- https://github.com/pschmitt/esphomeyaml-config/blob/master/config/sonoff_p0.yaml#L11

vs.

- https://github.com/pschmitt/esphomeyaml-config/blob/master/config/sonoff_p0.yaml#L63
- https://github.com/pschmitt/esphomeyaml-config/blob/master/config/sonoff_p0.yaml#L66

The linked configuration is working as expected[*] - but having to call once `turn_on` and then afterwards `turn_off` seems strange. Is the `inverted` stuff set late (after `50`)?

> [*] Intended behavior:
> - Have the mains relay power on on boot (just like `PowerOnState 1` does on Tasmota)
> - Sync the LED with the mains power switch: when on, turn on the LED, else turn off
> - A button press should toggle the mains

OttoWinter, 2018-10-10:
Ok, just so I understand correctly:

 * The relay *and* LED should be HIGH on boot.
 * The relay should display ON in hass when the output is HIGH.
 * The relay and light should always be synced. A LOW LED means a HIGH relay (LED `LOW` because it's on when it receives a digital HIGH).
 * Button should toggle both of them.

So, looking at your config, I don't think you need that many `invert`s (at least can't understand what's going on after 3 different inverts for the same stuff). So, would this configuration work?

```yaml
esphomeyaml:
  # ...
  on_boot:
    priority: 50
    then:
    - switch.turn_on: template_relay_1

binary_sensor:
- platform: gpio
  name: "Sonoff p0: Button"
  pin:
    number: GPIO0
    mode: INPUT_PULLUP
    inverted: true
  on_click:
    min_length: 50ms
    max_length: 350ms
    then:
    - switch.toggle: template_relay_1

switch:
- platform: gpio
  id: relay_1
  pin: GPIO23
- platform: template
  id: template_relay_1
  name: "Sonoff p0: Relay"
  optimistic: true
  turn_on_action:
    then:
    - switch.turn_on: relay_1
    - light.turn_on: led
  turn_off_action:
    then:
    - switch.turn_off: relay_1
    - light.turn_off: led

output:
- platform: esp8266_pwm
  id: basic_green_led
  inverted: True  # The LED is active-low
  pin: GPIO13

light:
- platform: monochromatic
  id: led
  output: basic_green_led
  default_transition_length: 0s
```

Basically, this configuration only exposes the template switch in the front-end. This is because everything should be in sync at all times anyway, so there's no real use in having multiple front-end switches for the same thing.

pschmitt, 2018-10-10:
Just tested your config (see [here for the exact config I used](https://github.com/pschmitt/esphomeyaml-config/blob/3b110588474088a18e7e1c168e60868f5833f9b0/config/sonoff_p0.yaml)). It won't turn on the LED nor the "mains" switch on boot, and the switch and LED are not in sync but in fact inverted. 

Mine was - albeit strange - working as "expected" for my Sonoff Basic.