esphome/esphome #160: update_interval slower that specified - Wemos D1 mini

DavidDeSloovere, 2018-10-11:
Don't know if this is the Wemos' fault or what's going on exactly, but I'm getting a consistent multiplication of the update_interval that is specified vs what is really running on the 'board'. It's like this both 1.8.1 and 1.8.2

I have to exact setups: a Wemos D1 mini with an ultrasonic sensor, both with a different interval
- `update_interval: 1s` is more like 15 seconds. (config, code, output below)
- `update_interval: 5s` turns out to be 75 seconds, which is 5 times the 15 seconds

Should I just change my config interval so that the actual interval is what I want? 
Can this be corrected in any way via config or esphomelib changes?
Is the Wemos not an ideal board for running actual stuff on and maybe just for prototyping? (although all boards are probably prototyping boards, but you might get what I'm trying to say)

The config (stripped down)

```
esphomeyaml:
  name: wemos_ultrasonic_01
  platform: ESP8266
  board: d1_mini

sensor:
  - platform: ultrasonic
    trigger_pin: D7
    echo_pin: D6
    name: "Ultrasonic Sensor"
    update_interval: 1s
```

The generated source code (stripped) - 1000ms passed as argument
```
// Auto generated code by esphomeyaml
#include "esphomelib/application.h"

using namespace esphomelib;

void setup() {
  App.set_name("wemos_ultrasonic_01");
  App.init_log(115200);

  ...

  Application::MakeUltrasonicSensor application_makeultrasonicsensor = App.make_ultrasonic_sensor("Ultrasonic Sensor", 13, 12, 1000);
  App.setup();
}

void loop() {
  App.loop();
  delay(16);
}

```

The output from esphomeyaml upload
```
INFO [esphomeyaml.espota] Waiting for result...
INFO [esphomeyaml.espota] Result: OK
INFO [esphomeyaml.__main__] Successfully uploaded program.
INFO [esphomeyaml.mqtt] Starting log output from wemos_ultrasonic_01/debug
[11:26:14][I][mqtt.client:137]: MQTT Connected!
[11:26:14][C][sensor.ultrasonic:032]: Setting up Ultrasonic Sensor...
[11:26:14][C][sensor.ultrasonic:033]:     Echo Pin: GPIO12
[11:26:14][C][esphal:067]:     GPIO Pin 12 with mode INPUT
[11:26:14][C][sensor.ultrasonic:035]:     Trigger Pin: GPIO13
[11:26:14][C][esphal:067]:     GPIO Pin 13 with mode OUTPUT
[11:26:14][C][sensor.ultrasonic:038]:     Pulse time: 10µs
[11:26:14][C][sensor.ultrasonic:039]:     Timeout: 11662µs
[11:26:14][C][component:243]:     Update interval: 1000ms
[11:26:14][C][sensor.mqtt:027]: Setting up MQTT Sensor 'Ultrasonic Sensor'...
[11:26:14][C][sensor.mqtt:029]:     Expire After: 45s
[11:26:14][C][sensor.mqtt:031]:     Unit of Measurement: 'm'
[11:26:14][C][sensor.mqtt:032]:     Accuracy Decimals: 2
[11:26:14][C][sensor.mqtt:033]:     Icon: 'mdi:arrow-expand-vertical'
[11:26:14][D][sensor.mqtt:093]: 'Ultrasonic Sensor': Pushing out value nan with 2 decimals of accuracy
[11:26:14][I][application:093]: Running through first loop()
[11:26:14][D][sensor.mqtt:093]: 'Ultrasonic Sensor': Pushing out value 0.214375 with 2 decimals of accuracy
[11:26:14][I][application:109]: First loop finished successfully!
[11:26:28][D][sensor.mqtt:093]: 'Ultrasonic Sensor': Pushing out value 0.214306 with 2 decimals of accuracy
[11:26:43][D][sensor.mqtt:093]: 'Ultrasonic Sensor': Pushing out value 0.214364 with 2 decimals of accuracy
[11:26:58][D][sensor.mqtt:093]: 'Ultrasonic Sensor': Pushing out value 0.214341 with 2 decimals of accuracy
[11:27:13][D][sensor.mqtt:093]: 'Ultrasonic Sensor': Pushing out value 0.214364 with 2 decimals of accuracy
[11:27:28][D][sensor.mqtt:093]: 'Ultrasonic Sensor': Pushing out value 0.214306 with 2 decimals of accuracy
[11:27:43][D][sensor.mqtt:093]: 'Ultrasonic Sensor': Pushing out value 0.214364 with 2 decimals of accuracy
[11:27:58][D][sensor.mqtt:093]: 'Ultrasonic Sensor': Pushing out value 0.214329 with 2 decimals of accuracy
[11:28:07][I][ota:143]: Boot seems successful, resetting boot loop counter.
[11:28:13][D][sensor.mqtt:093]: 'Ultrasonic Sensor': Pushing out value 0.214318 with 2 decimals of accuracy
[11:28:28][D][sensor.mqtt:093]: 'Ultrasonic Sensor': Pushing out value 0.214341 with 2 decimals of accuracy
```

OttoWinter, 2018-10-11:
Please see the "default filter" link next to the update interval option https://esphomelib.com/esphomeyaml/components/sensor/ultrasonic.html#configuration-variables :)

DavidDeSloovere, 2018-10-12:
Omg. How did I miss that. Spent some time 'debugging' this. Thanks.