esphome/esphome #170: What happened to use_build_flags?

snizzleorg, 2018-10-12:
Hi I updated to to 1.8.2 and if I try start the logs I get this:

```
INFO [esphomeyaml.config] Reading configuration...
Failed config
  esphomeyaml: 
    - Invalid config for [esphomeyaml]: [use_build_flags] is an invalid option for [esphomeyaml]. Check: esphomeyaml->use_build_flags. (See /config/esphomeyaml/buzzer.yaml, line 1). 
    - platform: ESP8266
      board: nodemcuv2
      name: buzzer
      use_build_flags: False
```

Has this option been removed?

OttoWinter, 2018-10-12:
It was replaced by 

```yaml
esphomeyaml:
  use_custom_code: True
```

because it will have a slightly different meaning in the future.

Apparently I missed the removal warning, adding it in #173

closed 2018-10-12 by OttoWinter

snizzleorg, 2018-10-12:
Great! Thanks!