esphome/esphome #186: Problem with delay

wuub, 2018-10-16:
**ESP (ESP32/ESP8266/Board/Sonoff):** board only, lolin32
**Affected component:** delay

**Description of problem:**
Delay does not seem to work as described. It does not stop the 

**Problem-relevant YAML-configuration entries:**
```yaml
switch:
  - platform: gpio
    pin: 16
    id: buzzer

binary_sensor:
  - platform: pn532
    uid: 06-8C-21-EC
    id: 'tag068C21EC'
    on_press:
      - mqtt.publish:
          topic: nfc01/nfc_tag/on_press
          payload: '06-8C-21-EC'
      - switch.turn_on: 
          id: buzzer
      - delay: 1s
      - switch.turn_off: 
          id: buzzer
```
 
**Traceback (if applicable):** n/a
**Additional information:**
buzzer sounds, but turns off immediately instead of waiting 1s.

wuub, 2018-10-16:
Changed the delay to 5s. Ran in VERY_VERBOSE mode

```
[21:00:17][V][pn532:083]: ATQA: 0x0004, SAK: 0x08
[21:00:17][V][mqtt.client:206]: Publish(topic='nfc01/nfc_tag/on_press' payload='B6-D7-20-EC' retain=0)
[21:00:17][VV][preferences:102]: '931d7fd-state' -> putting bool ON
[21:00:17][V][component:081]: set_timeout(name='', timeout=5000)
[21:00:17][VV][preferences:102]: '931d7fd-state' -> putting bool OFF
[21:00:17][V][component:081]: set_timeout(name='OFF', timeout=1500)
[21:00:17][VV][component:136]: Running interval 'update':0 with interval=1000 last_execution=3958 (now=4959)
```

OttoWinter, 2018-10-18:
Ok, so I think the problem is that you don't have a `then:` block in your configuration.

```yaml
binary_sensor:
  - platform: pn532
    uid: 06-8C-21-EC
    id: 'tag068C21EC'
    on_press:
      then:
        - mqtt.publish:
            topic: nfc01/nfc_tag/on_press
            payload: '06-8C-21-EC'
        - switch.turn_on: 
            id: buzzer
        - delay: 5s
        - switch.turn_off: 
            id: buzzer
```

```yaml
# esphomeyaml auto-converts it to this:

binary_sensor:
  - platform: pn532
    uid: 06-8C-21-EC
    id: 'tag068C21EC'
    on_press:
    - then:
        - mqtt.publish:
            topic: nfc01/nfc_tag/on_press
            payload: '06-8C-21-EC'
    - then:
        - switch.turn_on: 
            id: buzzer
    - then:
        - delay: 5s
    - then:
        - switch.turn_off: 
            id: buzzer
```

without it, esphomeyaml will interpret the configuration like it's many different automations to perform at once. I will try and see if I can make esphomeyaml read that config in a way that makes more sense.

wuub, 2018-10-18:
Oh, sorry, my mistake! 
It is a bit unexpected but 100% reasonable explanation, please feel free to close this issue :)

closed 2018-10-20 by OttoWinter