esphome/esphome #195: light.toggle using binary output component not working?

beireken, 2018-10-19:
I'm trying to toggle a binary light with a binary sensor on an esp32.
However esphomeyaml failed compiling with the error:
`src/main.cpp:45:47: error: 'class esphomelib::output::GPIOBinaryOutputComponent' has no member named 'make_toggle_action'`

I have the following configured:

```
output:
  - platform: gpio
    pin: 
      number: GPIO2
      inverted: True
    id: living1

light:
  - platform: binary
    name: "Living Room"
    output: living1

binary_sensor:
  - platform: gpio
    name: "Living Room"
    device_class: light
    pin:
      number: GPIO13
      inverted: True
      #mode: INPUT_PULLUP
    on_press:
      then:
        - light.toggle:
            id: living1
```

I think i'm doing something wrong?
At first i used the relay compontent with the binary sensor compontent to use switch.toggle. This worked fine. But i thought using the output and light components are how it should be done no?

OttoWinter, 2018-10-19:
esphomeyaml doesn't check if the thing you pass to `light.toggle` is actually a light (and it would be a ton of work to support that).

You're trying to toggle **the light**, not the **output**, right?

```yaml
light:
  - platform: binary
    name: "Living Room"
    output: living1
    id: light_1

binary_sensor:
  - platform: gpio
    name: "Living Room"
    device_class: light
    pin:
      number: GPIO13
      inverted: True
      #mode: INPUT_PULLUP
    on_press:
      then:
        - light.toggle: light_1
```

beireken, 2018-10-19:
oh, all i had to do was add an id to the light component and use that in the sensor then: part.
Thank a lot!

closed 2018-10-19 by beireken