esphome/esphome #203: Cannot be used if action in mqtt.on_message trigger

r-jordan, 2018-10-20:
**Operating environment (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP8266

**Affected component:**
https://esphomelib.com/esphomeyaml/components/mqtt.html#mqtt-on-message

**Description of problem:**
I would like to make some action depending on the payload of a mqtt message received.
When a conditional "if" action is used inside an mqtt.on_message trigger a compiling error is showed.

**Problem-relevant YAML-configuration entries:**
```yaml
mqtt:
  broker: 'xxx'
  username: 'xxx'
  password: 'xxx'
  discovery: False
  on_message:
    topic: devicex/ota_mode
    then:
      - if:
          lambda: 'return strcmp(x,"ON")==0;'
        then:
          - deep_sleep.prevent:
              id: deep_sleep_1
```

**Traceback (if applicable):**
```
Failed config
 mqtt:
   - Invalid config for [mqtt]: Cannot have two actions in one item. Key if overrides then! for dictionary value @ data['on_message']['then']. Got [OrderedDict([('if', OrderedDict([('lambda', 'return strcmp(x,"ON")==0;')])), ('then', [OrderedDict([('deep_sleep.prevent', OrderedDict([('id', 'deep_sleep_1')]))])]), ('action_id', None)])]. (See devicex.yaml, line 16). 
    - broker: xxx
      discovery: False
      on_message: [source devicex.yaml:21]
        then: ?
          - action_id: None
            if: [source devicex.yaml:24]
              lambda: return strcmp(x,"ON")==0;
            then: ?
              - deep_sleep.prevent: [source devicex.yaml:27]
                  id: deep_sleep_1
        topic: devicex/ota_mode
      password: xxx
      username: xxx
```

**Additional information:**

stoinov, 2018-10-24:
I am unable to make this one work:
```yaml
  on_message:
    topic: test8266/light/ledstrip/switch
    then:
      - if:
        condition:
          lambda: 'x == "ON";'
        then:
          - light.turn_on:
              id: ledstrip
              brightness: 100
        else:
          - light.turn_off: ledstrip

```

r-jordan, 2018-10-26:
It's compiling for me with this:

``lambda: 'return x == "ON";'``

For lights is more convenient to use any light component for example:

https://esphomelib.com/esphomeyaml/components/light/monochromatic.html

OttoWinter, 2018-10-26:
@stoinov Please see my comment from before, everything in the `if:` block needs to be indented

```yaml
  on_message:
    topic: test8266/light/ledstrip/switch
    then:
      - if:
          condition:
            lambda: 'x == "ON";'
          then:
            - light.turn_on:
                id: ledstrip
                brightness: 100
          else:
            - light.turn_off: ledstrip
```

stoinov, 2018-10-27:
I tried both suggestions and am still not able to. Here's the error:
```
  mqtt:
    - Invalid config for [mqtt]: value must be at most 1 for dictionary value @ data['on_message'][0]['then'][0]['if']['then'][0]['light.turn_on']['brightness']. Got None. (See .\test8266.yaml, line 11).
```

Line 11 is the `mqtt:` command. The version I'm using is 1.8.2

webcharlyx, 2018-11-20:
I have a problem here with Deep_sleep and on_message triggers:
Using the code from boverby, the state of d32_1/ota_mode and d32_1/led5 is not updated by the broker after waking up from deep_sleep.
Changes only take effect if they are sent during wake time.
Is this planned like this? 
In my opinion it would be helpful, if topics to which the node is linked by on_message triggers, automatically get updated on wakeup.

OttoWinter, 2018-11-20:
@webcharlyx https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages

webcharlyx, 2018-11-20:
works perfectly, THX!
sorry, the retained flag of mqtt messages was new to me.

OttoWinter, 2018-11-27:
Closing this issue because I believe this is fixed. Please comment if I'm mistaken.

closed 2018-11-27 by OttoWinter