esphome/esphome #204: FastLED SPI Light - effects in HA not working

try65do, 2018-10-21:
**Operating environment (Hass.io/Docker/pip/etc.):**
 Docker

**ESP (ESP32/ESP8266/Board/Sonoff):**
esp8266 / WeMos Di Mini

**Affected component:**

FastLED SPI Light

https://esphomelib.com/esphomeyaml/components/light/fastled_spi.html

**Description of problem:**

Effects not present in Home Assistant after auto discovery, all other aspects work as expected.

**Problem-relevant YAML-configuration entries:**
```yaml
light: 
  - platform: fastled_spi
    chipset: WS2801
    clock_pin: GPIO2
    data_pin: GPIO3
    name: "Kitchen Lights"
    num_leds: 58
    rgb_order: RGB
    effects:
      - fastled_color_wipe:
          name: Color Wipe Effect With Custom Values
          colors:
            - red: 100%
              green: 100%
              blue: 100%
              num_leds: 1
            - red: 0%
              green: 0%
              blue: 0%
              num_leds: 1
          add_led_interval: 100ms
          reverse: False
```

**Traceback (if applicable):**

```

```

**Additional information:**

Discovery message:
```json
{  
   "brightness":true,
   "rgb":true,
   "flash":true,
   "name":"Kitchen Lights",
   "platform":"mqtt_json",
   "state_topic":"kitchen_lights/light/kitchen_lights/state",
   "command_topic":"kitchen_lights/light/kitchen_lights/command",
   "availability_topic":"kitchen_lights/status"
}
```

OttoWinter, 2018-10-21:
Ok, two things:

Could you try with the beta I released yesterday? `docker pull ottowinter/esphomeyaml:rc` and `docker run [...] -it esphomeyaml:rc [...]` - One problem in 1.8.2 was that JSON messages would have a fixed maximum size and everything beyond that was cut off.

Also, everytime you add a new effect, you need to restart Home Assistant as it can't "re-discover" entities.

try65do, 2018-10-22:
No dice, I was only able to run the dev build as your above command  resulted in `ottowinter/esphomeyaml:rc not found`.

When attempting to run the dev build I get this error `src/main.cpp: In function 'void setup()':
src/main.cpp:10:7: error: 'class esphomelib::Application' has no member named 'set_compilation_datetime'
App.set_compilation_datetime(__DATE__ ", " __TIME__);
^
*** [.pioenvs/test/src/main.cpp.o] Error 1
`

Noted on the need to restart HA service, have been doing this each time.

try65do, 2018-11-02:
I noticed some changes to the repo, tried again with RC and it worked fine, was able to enable all effects and flash OTA - thanks, brilliant work.

closed 2018-11-02 by try65do