esphome/esphome #210: Can't build firmware

LucaP1, 2018-10-28:
Hello there,

I'm trying to use the esphomeyaml Add-On for Hass.io on my Raspberry Pi Zero W running HassOS, but no matter how my .yaml file looks or what board I've selected I got the following error.
The complete building log:

INFO [esphomeyaml.config] Reading configuration...
INFO [esphomeyaml.__main__] Generating C++ source...
INFO [esphomeyaml.__main__] Compiling app...
INFO [esphomeyaml.__main__] Running:  platformio run -d /config/esphomeyaml/wohnzimmerlichtschalter
Processing wohnzimmerlichtschalter (platform: espressif8266@>=1.8.0; board: esp01; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 512k
SYSTEM: ESP8266 80MHz 80KB RAM (512KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphomelib> 1.8.2 #c954c39
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP Async WebServer> 1.1.1
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <FastLED> 3.2.0 #d27bac9
|   |   |-- <EspSoftwareSerial> 3.4.1
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <Wire> 1.0
Building .pioenvs/wohnzimmerlichtschalter/firmware.bin
*** [.pioenvs/wohnzimmerlichtschalter/firmware.bin] Error -11
========================= [ERROR] Took 160.52 seconds =========================

Like I said I've tried diferent boards and configs but none worked.
I'm running all the latest updates and even tried the beta and dev versions but that didn't changed anything.
Maybe I'm missing out something but I really dont know what the problem could be.
The atteched log corresponds to the folowing .yaml configuration but I don't have any device configured.

esphomeyaml:
  name: wohnzimmerlichtschalter
  platform: ESP8266
  board: esp01

wifi:
  ssid: 'MY_SSID'
  password: 'MY_WIFI_PASSWORD'

mqtt:
  broker: 'MY_MQTT_URL'
  username: 'MY_MQTT_USERNAME'
  password: 'MY_MQTT_PASSWORD'

# Enable logging
logger:

ota:
  password: 'MY_OTA_PASSWORT'

Greetings Luca.

OttoWinter, 2018-11-08:
Most likely this is due to the machine HassIO is running not having enough memory.

The error is `-11` which is a segfault due to too little memory. That also seems plausible since running HassIO, Home Assistant, (other add-ons), esphomeyaml on a single tiny RPi Zero W is likely to have very little remaining memory. And the compiler `gcc` is known to use up a lot of memory. Since `gcc` is the only compiler espressif provides, this can't be fixed.

roadfox, 2018-11-18:
you can try to change the memory splitting and add 16MB (i thinks it's the minimum) to the graphics card in raspi-config.
also you can add a swap partition if not done so already to free up some more memory. 
but i'm not shure if this will help

OttoWinter, 2018-11-27:
I don't know much about swap files and this stuff. Anyway, it's not really an esphomeyaml issue and more of a HassOS issue. Closing this here...

closed 2018-11-27 by OttoWinter