esphome/esphome #212: Fallback Wifi Network(s)

pyrosmiley, 2018-10-29:
**Is your feature request related to a problem? Please describe.**
I have a main wifi network and an extension network to reach the areas where my main one does not reach. Unfortunately, there are spots where both signals are spotty, and it's a crapshoot as to whether Network A or Network B is currently stronger at any given moment. 

Also, I like to travel with a small travel router, but in order to bring my DIY alarm clock with me, I need to flash new firmware onto two ESPs just so to update the wifi network. Keeping them on the separate network at home (as a hotspot) isn't super feasible because mDNS fails, preventing OTA updates.

**Describe the solution you'd like**
Ideally, I'd like to specify multiple wifi networks in the configuration; when booting, esphomelib would scan for the strongest of the specified networks and select that one. Alternatively (or even in addition, with a flag in the config!) the ESP would simply try to connect to each network specified, in order, and only when all options fail would it timeout and reset or start the AP. 

**Additional context**
For reference of what I'm thinking of, I really like the way ESPurna implements this one particular thing -- I can specify a few networks, and my stuff still works even when I bring it with me, or one of the signals at home is weaker than it should be.

roadfox, 2018-11-18:
for the wifi's you own, just set them all to the same ssid and psk but on different channels. all halfway modern wifi stacks should select the strongest channel and try to connect.

OttoWinter, 2018-11-27:
Tracking this feature here: https://github.com/OttoWinter/esphomelib/issues/49 - I hope I can get it done for 1.10.0 (including WiFi provisioning)

closed 2018-11-27 by OttoWinter