esphome/esphome #213: Dallas component on "wemos mini pro" pin D3 causes reset

kenci, 2018-10-30:
**Operating environment (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266/Board/Sonoff):**

**Affected component:**

**Description of problem:**
When using D3 pin for dallas component it repeatedly crashes the esp8266. When i am using e.g D6 it's working without any problems.

**Problem-relevant YAML-configuration entries:**
```yaml
dallas:
  - pin: D3
```

**Traceback (if applicable):**

```
[13:23:26][W][ota:124]: Last Boot was an unhandled reset, will proceed to safe mode in 1 restarts
[13:23:26][C][sensor.dallas:044]: Setting up DallasComponent...
[13:23:26][C][sensor.dallas:045]:     Want device count: 0
[13:23:26]
[13:23:26]Exception (29):
[13:23:26]epc1=0x40206110 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000
[13:23:26]
[13:23:26]ctx: cont
[13:23:26]sp: 3ffffbf0 end: 3fffffd0 offset: 01a0
[13:23:26]
[13:23:26]>>>stack>>>
[13:23:26]3ffffd90:  00000014 3fff2ba4 3ffe9bd4 4020742d
[13:23:26]3ffffda0:  3ffffdf0 3ffffde0 0000000c 00000000
[13:23:26]3ffffdb0:  3ffffdf0 3ffffde0 0000000c 00000000
[13:23:26]3ffffdc0:  00000000 00000002 401066d8 40207450
[13:23:26]3ffffdd0:  3ffffdf0 3ffffde0 3ffe854c 3fff2ba4
[13:23:26]3ffffde0:  00000014 00000050 4020c234 3fffefb0
[13:23:26]3ffffdf0:  0000002d 00000000 3ffe9bd4 4020c2ee
[13:23:26]3ffffe00:  00000000 3fff2ba4 3ffe9bd4 40209ffe
[13:23:26]3ffffe10:  00000000 00000000 4bc6a7f0 00000000
[13:23:26]3ffffe20:  00000000 00000000 401066d8 00001bde
[13:23:26]3ffffe30:  00000000 00000001 3ffe854c 3fff265c
[13:23:26]3ffffe40:  00203952 00000102 4020c234 00000050
[13:23:26]3ffffe50:  00000014 3fff2ba4 3fff2ba4 40205c94
[13:23:26]3ffffe60:  00000007 00000102 4020c234 3fffefb0
[13:23:26]3ffffe70:  00000007 3fff265c 3fff0ec8 402057f0
[13:23:26]3ffffe80:  00000007 3fff2ba4 3fff0ec8 40204b0e
[13:23:26]3ffffe90:  3fff39fc 00000008 3fff0ec8 401006dc
[13:23:26]3ffffea0:  00000008 00000100 0000001c 3fff3084
[13:23:26]3ffffeb0:  00000000 00000000 00000000 3fff32a4
[13:23:26]3ffffec0:  3fffff44 3fff3144 3fff0ec8 40203672
[13:23:26]3ffffed0:  3fff32ec 3fff32f4 3fff32f4 3fff31cc
[13:23:26]3ffffee0:  3fff3204 3fff3214 3fff324c 3fff325c
[13:23:26]3ffffef0:  3fff3294 3fff32a4 3fff32dc 00000000
[13:23:26]3fffff00:  40214290 00000000 3fffff44 00000002
[13:23:26]3fffff10:  3fff3318 3ffffed0 0000000b 3fffff14
[13:23:26]3fffff20:  00000001 3ffffed0 00000007 3fffff08
[13:23:26]3fffff30:  00000003 feefeffe feefeffe feefeffe
[13:23:26]3fffff40:  feefeffe 3fff2eec 3fff3144 3fff2de4
[13:23:26]3fffff50:  feefeffe 3fff2cbc feefeffe feefeffe
[13:23:26]3fffff60:  3fff2c54 3fff2bd4 3fff324c 3fff2eec
[13:23:26]3fffff70:  3fff2de4 3fff2f24 3fff2f14 3fff3294
[13:23:26]3fffff80:  3fff325c 3fff3204 3fff31bc 3fff3214
[13:23:26]3fffff90:  3fff3184 3fff3174 3fff31cc feefeffe
[13:23:26]3fffffa0:  feefeffe feefeffe feefeffe 3fff1034
[13:23:26]3fffffb0:  3fffdad0 00000000 3fff102c 4020c2b8
[13:23:26]3fffffc0:  feefeffe feefeffe 3ffe854c 40100721
[13:23:26]<<<stack<<<
[13:23:26]
[13:23:26] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[13:23:26]
[13:23:27]load 0x4010f000, len 1384, room 16
[13:23:27]tail 8
[13:23:27]chksum 0x2d
[13:23:27]csum 0x2d
[13:23:27]vbb28d4a3
[13:23:27]~ld
```

**Additional information:**

OttoWinter, 2018-11-02:
That pin maps to GPIO0,. which is often required by the bootloader. However, at runtime it should work.

Code-wise I can't see any reason why that pin specifically would not work, so I will need to decode the stack trace to find out more about the bug. Fortunately, 1.9.0 will have such an "automatic stack trace decoder, I'll look at this issue again then.

mezz64, 2018-12-27:
I've noticed an odd bit of behavior with pin D3 on a Wemos mini and the syntax used in the code.  If you specify "D3" as the pin in the yaml the generated code file gets changed to "0".  In this scenario you get the crash as described by @kenci.  However, if you manually compile the .cpp with the pin specified as D3, not 0, using vsCode everything works as expected with no crash.

So something funky is going on between the definition change from D3 to 0, even though they should both do the same thing.

mezz64, 2019-01-12:
@OttoWinter It's a generic Wemos D1 mini v3 board that only flashes successfully as a nodemcuv2.

closed 2019-02-02 by OttoWinter