esphome/esphome #216: Binary sensor multiple on_click commands?

beireken, 2018-11-03:
My use case is as follows:
One binary sensor (wall push button) with functions needed:
- single short press, light 1 toggle
- double press, light 2 toggle
- long single press (eg. 2.5 seconds), lights 1,2, ... off

I have the following code (with added extra on_click function) that does not compile.
`binary_sensor:
  - platform: gpio
    name: "Living Room"
    device_class: light
    pin:
      number: GPIO13
      inverted: True
      mode: INPUT_PULLUP
    on_click:
      min_length: 130ms
      max_length: 1000ms
      then:
        - light.toggle:
            id: light_keuken
    on_double_click:
      min_length: 50ms
      max_length: 350ms
      then:
        - light.toggle:
            id: light_living1
    on_click:
      min_length: 2000ms
      max_length: 3000ms
      then:
        - light.turn_off:
            id: light_keuken     
        - light.turn_off:
            id: light_living1
        - light.turn_off:
            id: light_berging
        - light.turn_off:
            id: light_bureau `

How can i configure this use case?

beireken, 2018-11-07:
is there a way we can ask questions / contact that aren't really issues?

brandond, 2018-11-07:
There's a link to Discord in the docs, I would suggest you take it there.
https://esphomelib.com/esphomeyaml/

closed 2018-11-09 by OttoWinter