esphome/esphome #217: Change WIFI access

voicevon, 2018-11-04:
I got many esp nodes working in my home. now I want to change my password of my router.I mean the Wifi password, not the admin password. But problem came, All the password in the nodes are hard coded. If I  use OTA to update the Wifi connection, 
1) I have to have two routers, one is the original, the other one is the new router.
2) The HASSIO must   connect to the old router, for node's OTA.
3) After the #1 esp node is updated,if I want to verify it, the HASSIO must connect to the new router.'
4) After verification is OK, I connect HASSIO back to the old router, for updating #2 esp node.
   This essentially is a repeating from step 2.

Anybody got better idea? Is possible to archive it like below?
1) I remove the old router.
2) Esp node lost connection to old router(that is the all router for version below Version 1.9 or Version 2.0).
**3) Esp node becomes an AP, and web server is enabled.**
4) My phone's Wifi connected to Esp node ,with wifi_Id and wifi_password is hard coded.
5) I open the web browser,and access "http://192.168.1.1"
6) I got an simple web UI, it let me select Wifi_AP, and input Wifi password.
**-- Esp node write this configuration to somewhere,EEPROM?**
7) I click the "Restart Esp node" button on web UI.
-**- Esp node connect Wifi-AP with the configuration, it's not hard coded.**
8) Esp_AP is disappeared, My phone's Wifi connect to the previous Wifi AP, Normally it's my router.
9) I can setup the #2 esp node.

OttoWinter, 2018-11-04:
esphomelib won't support having the WiFi configuration in the EEPROM, as using the Flash/EEPROM directly causes all sorts of issues with ESPs (esphomelib had something like this before, but there were so many edge cases with different boards).

But, you can set up esphomelib to create an AP if the WiFi network disappears: https://esphomelib.com/esphomeyaml/components/wifi.html#access-point-mode

I think the best way for you would be:

 * Connect HassIO to old router
 * For each node:
   * Update YAML configuration of node to point to new router
   * Compile & upload
   * Verify everything works by connecting phone to new router and checking if webinterface shows up
 * Connect HassIO to new router

voicevon, 2018-11-05:
Yes, That is a better way. the drawback is I have to get the original yaml file. So I hope another feature that the node  can save/backup the yaml file of its code. Currently my idea is not clearly, it should be like:
1) The node use spiffs to store the yaml file.
2) There is a yaml_text_sensor, so we can read out the yaml file.
3) When we update firmware by OTA, the yaml file is updated(sync) also.  
I don't know how difficult of implementation, seems esphomelib has no reference to SPIFFS right now.

voicevon, 2018-11-07:
I don't know whether Platformio support SPIFFS or not. if yes, then it should be easy to store yaml file to SPIFFS inside the node. In setup() of   Esphomelib main.cpp, can read configuration from SPIFFS.
By my experience, There should be two files in SPIFFS, first is the ymal file, the second is a configuration file, include properties of  wifi,mqtt,etc. 

The result in esphomeyaml is like:
1)There is no *.Yaml file in HassIO(or the file is just a backup). the file is in the node.
2) Can read out myNode.Yaml over the air.
3)  When I press "UPLOAD", OTA will update both myNode.bin, and myNode.yaml file.

OttoWinter, 2018-11-08:
As I said, I don't like storing things in flash (+ you can search the esphomelib repo for other long explanations for this).

But primarily, I'd like to question your use-case. The thing you want to do is called "creating backups". Just open the file, copy the contents into a new file on your computer and restore if needed. Why should esphomelib implement this just for this small use-case which can **easily** be achieved otherwise through backups.

Also, SPIFFS just takes up more flash space, which the ESPs already lack a lot. 

Yes, changing the WiFi network for esphomelib nodes is difficult, but esphomelib is stateless and the only way this process will improve is when esphomelib gets the option to define an alternative WiFi network to connect to when the primary one is down.

closed 2018-11-08 by OttoWinter