esphome/esphome #219: ADC: 'id(my_sensor).update();' does not compile

webcharlyx, 2018-11-07:
**Operating environment (Hass.io/Docker/pip/etc.):**
VScode, pip, platformio-dev

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP-WROOM-32 (wemos)

**Affected component:**

https://esphomelib.com/esphomeyaml/components/sensor/adc.html

**Description of problem:**
Within a Lambda: A manual ADC value update does not compile on error: 'class esphomelib::sensor::Sensor' has no member named 'update'

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  # ...
  on_boot:
    priority: -10
    # ...
    then:
      - output.turn_on: GPIO_con0
      - delay: 100ms
      - lambda: 'id(ADC_con0).update();'
      - output.turn_off: GPIO_con0

sensor:
  - platform: adc
    pin: GPIO36
    name: "ADC_con0"
    attenuation: 11db
    id: ADC_con0

output:
  - platform: gpio
    pin: 27
    inverted: False
    id: GPIO_con0
```

**Traceback (if applicable):**

```
PS > esphomeyaml .\ESP32_Keypanel.yaml run
?[32mINFO [esphomeyaml.config] Reading configuration...?[0m
?[32mINFO [esphomeyaml.__main__] Generating C++ source...?[0m
?[32mINFO [esphomeyaml.__main__] Compiling app...?[0m
?[32mINFO [esphomeyaml.__main__] Running:  platformio run -d '.\esp_keypanel'?[0m
Processing esp_keypanel (platform: espressif32@>=1.4.0; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
SYSTEM: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoOTA> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <Update> 1.0
|-- <Wire> 1.0
|-- <esphomelib> 1.8.2 #c954c39
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <AsyncTCP> 1.0.1
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP Async WebServer> 1.1.1
|   |   |-- <AsyncTCP> 1.0.1
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <FastLED> 3.2.0 #d27bac9
|   |-- <WiFi> 1.0
|   |-- <Preferences> 1.0
|   |-- <Wire> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESPmDNS> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <Update> 1.0
Compiling .pioenvs\esp_keypanel\src\main.cpp.o
src\main.cpp: In lambda function:
src\main.cpp:256:17: error: 'class esphomelib::sensor::Sensor' has no member named 'update'
ADC_con0->update();
^
*** [.pioenvs\esp_keypanel\src\main.cpp.o] Error 1
================================================================================== [ERROR] Took 5.95 seconds ==================================================================================
```

**Additional information:**

brandond, 2018-11-07:
What about `id(ADC_con0).adc.update();`. id() just gives you the sensor, you want the actual ADC that it's wrapping. Check the reference docs here:
https://esphomelib.com/api/sensor/adc.html

OttoWinter, 2018-11-08:
@webcharlyx In 1.9.0 there will be a new `component.update` action, but I added some better typing here: https://github.com/OttoWinter/esphomeyaml/pull/225

closed 2018-11-08 by OttoWinter