esphome/esphome #228: Remote receiver/transmitter does not work properly with Samsung Smart TV

korvad, 2018-11-12:
Hi, @OttoWinter 
First of all thx for great library. It really impressive.

**Operating environment (Hass.io/Docker/pip/etc.):**
**Rapberry Pi 3, python2.7, pip 18.1, Home Assistant 0.76.2**

**ESP (ESP32/ESP8266/Board/Sonoff):**
**NodeMCU v3 Lolin**

**Affected component:**
https://esphomelib.com/esphomeyaml/components/remote_receiver.html
https://esphomelib.com/esphomeyaml/components/binary_sensor/remote_receiver.html
https://esphomelib.com/esphomeyaml/components/remote_transmitter.html
https://esphomelib.com/esphomeyaml/components/switch/remote_transmitter.html

**Description of problem:**
I'm trying to migrate my arduino IR switch for samsung smart TV
to esphomelib using remote receiver/transmitter component but without success yet.
All the stuff are working cause with arduino sketch IRremote i can successfully dump
all needed codes for my samsung TV and then send them.
E.g:
0xE0E040BF - power on/off
0xE0E0F00F - mute
0xE0E048B7 - channel up
0xE0E008F7 - channel down
etc.

then it send througth
   irsend.sendRaw({0xE0E040BF}, 1, 38);
for example and it works fine

The problem is that i can not pass samsung values dumped through IRremote
with esphomelib - there is not description in documentation for this.
So my guess was to use raw values but when I'm trying to dump raw values from receiver then every time i press the button the new raw code sequence differs from the previous one
E.g:
[0;36m[D][remote.raw:051]: Received Raw: -4457, 4482, -602, 1648, -601, 1649, -577, 1676, -604, 533, -591, 527, -572, 554, -602, 526, -595, 527, -595, 1647, -578, 1673, -601, 1653, -578, 556, -591, 528, -598, 528, -595, 529, -599, 544, -595, 1649, -601, 1648, -602, 1639, -614, [0m
[0;36m[D][remote.raw:064]:   526, -575, 556, -593, 527, -598, 527, -598, 528, -598, 530, -595, 530, -596, 527, -595, 1649, -600, 1649, -601, 1649, -576, 1674, -602, 1651, -600, 10000
[0;36m[D][remote.raw:051]: Received Raw: -4464, 4488, -582, 1671, -599, 1654, -595, 1657, -595, 531, -571, 608, -515, 555, -596, 532, -567, 554, -570, 1679, -569, 1679, -598, 1654, -599, 530, -570, 557, -592, 532, -594, 532, -569, 569, -595, 1705, -545, 1652, -595, 1657, -598, [0m
[0;36m[D][remote.raw:064]:   534, -589, 536, -568, 558, -592, 531, -570, 558, -567, 556, -570, 606, -544, 532, -591, 1665, -584, 1653, -597, 1655, -598, 1651, -600, 1652, -597, 10000
...

I had used 2 different receivers with same result.
So I checked the code and found that esphomeyaml greater of 1.8.2 has support for samsung decoding
I have checkout it out and install, then i used samsung dump specially but in this case there is no codes in serial log.

So it's not clear for me how i can use remote receiver/sender component for my case properly.

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: kitchen_ir_switch
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 'ssiid'
  password: 'password'
  manual_ip:
    static_ip: 192.168.0.113
    gateway: 192.168.0.1
    subnet: 255.255.255.0

mqtt:
  broker: 'ip'
  username: 'username'
  password: 'password'

logger:
  level: DEBUG

remote_receiver:
  pin: D6
  dump: all

remote_transmitter:
   pin: D2
   carrier_duty_percent: 50%

switch:
  - platform: remote_transmitter
    name: "samsung_tv_ir_switch"
    raw:
      carrier_frequency: 38kHz
      data: [4500, 4400, 600, 1650, 600, 1600, 600, 1650, 600, 500, 600, 500, 600, 550, 600, 500, 600, 500, 600, 1650, 600, 1600, 600, 1600, 650, 500, 600, 500, 600, 550, 600, 500, 600, 500, 600, 550, 600, 1600, 600, 500, 650, 500, 600, 500, 600, 500, 600, 550, 600, 500, 600, 1600, 600, 550, 600, 1600, 600, 1650, 600, 1600, 600, 1650, 600, 1600, 600, 1650, 600]
ota:

```

**Traceback (if applicable):**
```
[0;35m[C][ota:445]: There have been 0 suspected unsuccessful boot attempts.[0m
[0;32m[I][application:053]: Running through setup()...[0m
[0;35m[C][wifi:027]: Setting up WiFi...[0m
[0;32m[I][wifi:276]: WiFi Connecting to 'arch_wireless'...[0m
[0;35m[C][wifi:293]:     STA Hostname: 'kitchen_ir_switch'[0m
[0;32m[I][wifi:324]: WiFi connected![0m
[0;35m[C][wifi:326]:     SSID: arch_wireless[0m
[0;35m[C][wifi:328]:     BSSID: B8:A3:86:61:7E:96[0m
[0;35m[C][wifi:329]:     Channel: 5[0m
[0;35m[C][wifi:330]:     IP Address: 192.168.0.113[0m
[0;35m[C][wifi:331]:     Subnet: 255.255.255.0[0m
[0;35m[C][wifi:332]:     Gateway: 192.168.0.1[0m
[0;35m[C][wifi:333]:     DNS1: 208.67.222.222[0m
[0;35m[C][wifi:334]:     DNS2: 0.0.0.0[0m
[0;35m[C][ota:032]: Setting up OTA...[0m
[0;35m[C][ota:033]:     port: 8266[0m
[0;35m[C][mqtt.client:020]: Setting up MQTT...[0m
[0;35m[C][mqtt.client:021]:     Server Address: 192.168.0.103:1883[0m
[0;35m[C][mqtt.client:022]:     Username: 'username'[0m
[0;35m[C][mqtt.client:023]:     Password: 'password'[0m
[0;35m[C][mqtt.client:026]:     Client ID: 'kitchen_ir_switch-807d3a675182'[0m
[0;35m[C][mqtt.client:028]:     Discovery prefix: 'homeassistant'[0m
[0;35m[C][mqtt.client:029]:     Discovery retain: true[0m
[0;32m[I][mqtt.client:097]: Connecting to MQTT...[0m
[0;32m[I][mqtt.client:133]: MQTT Connected![0m
[0;35m[C][remote.receiver:289]: Setting up Remote Receiver...[0m
[0;35m[C][esphal:059]:     GPIO Pin 12 with mode INPUT[0m
[0;33m[W][remote.receiver:299]: Remote Receiver Signal starts with a HIGH value. Usually this means you have to invert the signal using 'inverted: True' ![0m
[0;35m[C][esphal:059]:     GPIO Pin 4 with mode OUTPUT[0m
[0;35m[C][switch.mqtt:023]: Setting up MQTT switch 'samsung_tv_ir_switch'[0m
[0;35m[C][switch.mqtt:024]:     Icon: ''[0m
[0;36m[D][mqtt.client:175]: Subscribing to topic='kitchen_ir_switch/switch/samsung_tv_ir_switch/command' qos=0...[0m
[0;36m[D][switch.mqtt:074]: 'samsung_tv_ir_switch': Sending state OFF[0m
[0;32m[I][application:092]: You're running esphomelib v1.9.0b4 compiled on Nov 12 2018, 23:35:54[0m
[0;32m[I][application:101]: Running through first loop()[0m
[0;32m[I][application:118]: First loop finished successfully![0m
```

escoand, 2018-11-12:
Have you seen this message in the logs?

`Remote Receiver Signal starts with a HIGH value. Usually this means you have to invert the signal using 'inverted: True' !`

yawor, 2018-11-20:
Use:
```yaml
remote_receiver:
  pin:
    number: D6
    inverted: yes
    mode: INPUT_PULLUP
  dump: all
```

closed 2018-11-27 by OttoWinter