esphome/esphome #231: New integration request: PZEM-004

rradar, 2018-11-13:
**What new integration would you wish to have?**
- Peacefair PZEM-004
- Peacefair PZEM-004T

**If possible, provide a link to an existing library for the integration:**
- https://halfforked.wordpress.com/2018/05/18/low-cost-house-power-monitoring-with-pzem-004-for-home-assistant/
- http://pdacontrolen.com/electricity-consumption-meter-peacefair-pzem-004-esp8266-arduino-nano/
- https://www.instructables.com/id/Power-Peacefair-PZEM-004-ESP8266-Arduino-Nano/

**Product link:**
- https://www.aliexpress.com/item/ANENG-4in1-Digital-Ammeter-Voltmeter-Watt-Power-Energy-Meter-AC-80-260V-100A-PZEM-004/32847092493.html

**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I can't use esphomelib for my integrations :cry: 

**Additional context**
Want to measure the total energy consumption/load of a one-phase main connection in real time with esphomelib/yaml :+1:

kvvoff, 2018-11-18:
+

ittchmh, 2019-01-16:
Hello, I want to test this integration
How can I do that? I see changes in PR but I didn't find information on how to do custom build of esphomelib + esphomeyaml
I have NodeMCU with ESPurna firmware, I want to replace firmware to esphomeyaml

ittchmh, 2019-01-16:
Ok, I have found how to use local repo (https://esphomelib.com/esphomeyaml/components/esphomeyaml.html#esphomeyaml-esphomelib-version)
Will try this PR

kvvoff, 2019-01-16:
@ittchmh 
If you succeed, please write how you did it. I am also interested in the possibility of connecting the pzem module. Thanks

OttoWinter, 2019-01-16:
@ittchmh @kvvoff That integration isn't finished (WIP). If I remember correctly it is not even reporting values in the current state.

ittchmh, 2019-01-16:
Ok, if it is not working at all I'll try custom module

I ready to test it when integration finished