esphome/esphome #232: FastLED Lambda Effect does not compile

guardmedia, 2018-11-15:
**Operating environment (Hass.io/Docker/pip/etc.):**

Ubuntu 18.04.1 LTS on Windows 10 WSL (Winows Subsystem Linux)
Linux version 4.4.0-17134-Microsoft (Microsoft@Microsoft.com) (gcc version 5.4.0 (GCC) ) #345-Microsoft Wed Sep 19 17:47:00 PST 2018
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)
gcc (Ubuntu 7.3.0-16ubuntu3) 7.3.0
(...not using Hass.io or Docker...)

**ESP (ESP32/ESP8266/Board/Sonoff):**

ESP32 (mhetesp32devkit)

**Affected component:**

"FastLED Clockless Light" and "Lambda Effect"

- https://esphomelib.com/esphomeyaml/components/light/fastled_clockless.html
- https://esphomelib.com/esphomeyaml/components/light/index.html?highlight=effect#lambda-effect

**Description of problem:**
When attempting to use lambda effects with FastLED, compilation fails due to an error stating `src/main.cpp:20:59: error: 'FastLEDLightOutputComponent' has not been declared`. Please see "Additional information" for proposed solution.

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: fastled
  platform: ESP32
  board: mhetesp32devkit

wifi:
  ssid: 'TotallyOriginalName'
  password: 'ReallyGreatPassword'
  id: my_wifi

light:
  - platform: fastled_clockless
    id: light_fastled
    chipset: WS2812B
    pin: GPIO21
    num_leds: 900
    rgb_order: GRB
    name: 'LED Strip'
    effects:
      - fastled_lambda:
          name: DemoReel
          update_interval: 8ms
          lambda: |-
            static uint8_t gCurrentPatternNumber = 0; // Index number of which pattern is current
            static uint8_t gHue = 0; // rotating "base color" used by many of the patterns
            
            static unsigned long nChangeHue = millis();
            if (millis() > nChangeHue) {
              nChangeHue += 20;
              gHue++;
            }
            
            static unsigned long nChangePattern = millis();
            if (millis() > nChangePattern) {
              nChangePattern += 10*1000; // Every 10 seconds
              gCurrentPatternNumber = (gCurrentPatternNumber + 1) % 5;
            }
            
            switch(gCurrentPatternNumber) {
              case 0: {
                ESP_LOGD("DemoReel", "rainbow() - FastLED's built-in rainbow generator");
                fill_rainbow(it.leds(), it.size(), gHue, 7);
                }; break;
              case 1: {
                ESP_LOGD("DemoReel", "rainbowWithGlitter() - built-in FastLED rainbow, plus some random sparkly glitter");
                fract8 chanceOfGlitter = 80;
[... 313 lines cut ...]
.piolibdeps/FastLED/platforms/esp/32/clockless_esp32.h:375:13: warning: unused variable 'cur' [-Wunused-variable]
int cur = 0;
^
*** [.pioenvs/fastled/src/main.cpp.o] Error 1
====================================================================================== [ERROR] Took 5.13 seconds ======================================================================================
```

**Additional information:**
By modifying adding `light::` class before calling `FastLEDLightOutputComponent`, the same YAML script compiles and the effect is able to run as expected.

```diff
diff --git a/esphomeyaml/components/light/__init__.py b/esphomeyaml/components/light/__init__.py
index c74376b..dbf2c9a 100644
--- a/esphomeyaml/components/light/__init__.py
+++ b/esphomeyaml/components/light/__init__.py
@@ -250,7 +250,7 @@ def build_effect(full_config):
         yield effect
     elif key == CONF_FASTLED_LAMBDA:
         lambda_ = None
-        args = [(RawExpression('FastLEDLightOutputComponent &'), 'it')]
+       args = [(RawExpression('light::FastLEDLightOutputComponent &'), 'it')]
         for lambda_ in process_lambda(config[CONF_LAMBDA], args):
             yield None
         yield FastLEDLambdaLightEffect.new(config[CONF_NAME], lambda_, config[CONF_UPDATE_INTERVAL])
```

OttoWinter, 2018-11-15:
There's a reason it's an undocumented feature :D - I still want to change the API a bit (see https://github.com/OttoWinter/esphomelib/pull/243) But thanks for the fix!

closed 2018-11-15 by OttoWinter

MattW2, 2019-06-24:
@guardmedia by any chance have you been able to get your example code with the `...switch(gCurrentPatternNumber) {...` lambda effect to work under the new `addressable_lambda:` effect. I have found many examples of using the style of FastLED library function calls under the previously named `fastled_lambda ` effect type but very few examples of addressable LED code ported over to the `addressable_lambda:` effect type. I am specifically trying to get the [FastLED XYMatrix](https://github.com/FastLED/FastLED/blob/master/examples/XYMatrix/XYMatrix.ino) example to work under ESPHome.

commit f394968b (2018-11-15): Fix fastled lambda light effect

Fixes #232
commit a169d375 (2018-11-25): Fix fastled lambda light effect

Fixes #232