esphome/esphome #233: Power_supply component not working with monochromatic light and fastled

the-skevenger, 2018-11-15:
Hello first i want to thank you for this very useful addon for home assistant!!!

But I have a issue with power supply component. I am using ws2812b strip and 12V warm white strip on one esp.
These two strips are powered by same ATX PSU. ESP32 devkit is connected directly to ATX 5V standby. White strip is connected to 12V with NPN tranzistor on ground and ws2812b is connected to 5V.
There was no problem with compilation and upload. When i turn on ws2812b from frontend or capacitive button its turn on ATX PSU (by pulling green wire low) correctly. But when i want turn on only white strip power_supply component is not pulling green wire low and ATX stays in standby mode.
White strip is powered only when ws2812b is turned on. When I turn off ws2812b ATX go to standby (in 10seconds) but white strip is still turned on in frontend.

Iam running esphomeyaml 1.9.0b1 and this is my code:

```yaml

esphomeyaml:
  name: lamp
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: '****'
  password: '****'
  
  manual_ip:
    static_ip: ***
    gateway: ***
    subnet: ***

mqtt:
  broker: '***'
  username: '***'
  password: '***'

logger: # Enable logging

ota:

light:
  - platform: fastled_clockless
    chipset: WS2812b
    power_supply: power_supply_id
    pin: GPIO05
    num_leds: 300
    rgb_order: GRB
    name: "RGB"
    id: RGB
          
  - platform: monochromatic
    name: "white"
    output: pwmoutput
    id: white

esp32_touch:
  setup_mode: false

binary_sensor:
  - platform: esp32_touch
    name: "Touch Pad"
    pin: GPIO14
    threshold: 1000
    on_click:
      min_length: 500ms
      max_length: 2000ms
      then:
        - light.toggle:
            id: white
    on_double_click:
      min_length: 0ms
      max_length: 500ms
      then:
        - light.toggle:
            id: RGB
    
output:
  - platform: ledc
    power_supply: power_supply_id
    pin: GPIO27
    id: pwmoutput
    
    
power_supply:
  - id: power_supply_id
    pin:
      number: GPIO25
      inverted: true

```

Maybe Iam doing something wrong but I tried more options and this code seems to be right.

Thanks for reply :)

OttoWinter, 2018-11-15:
Try with 1.9.0 (which I released today 😉). That bug should be fixed there

the-skevenger, 2018-11-15:
awesome realy thanks :)

closed 2018-11-15 by the-skevenger