esphome/esphome #236: Can't compile in Version 1.9 (1.8.x was working)

rradar, 2018-11-17:
**Operating environment (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266/Board/Sonoff):**

esp8285 (esp m3)

**Affected component:**

not sure

**Description of problem:**

Can't compile (and upload) my sketch anymore in version 1.9. Did touch/change my yaml. Compiling with version 1.8.2 was working as expected.

**Problem-relevant YAML-configuration entries:**
```
i2c:
  sda: 0
  scl: 2
  scan: False

binary_sensor:
  - platform: status
    name: "ESPM3 Status"
    id: status

sensor:
  - platform: wifi_signal
    name: "ESPM3 WiFi Signal"
    update_interval: 300s
    id: signal

  - platform: bmp085
    temperature:
      name: "ESPM3 Temperature"
      id: temperature
    pressure:
      name: "ESPM3 Pressure"
      id: pressure
    update_interval: 300s

font:
  - file: "five.ttf"
    id: five8
    size: 8

time:
  - platform: sntp
    id: time
    servers:
      - 0.pool.ntp.org

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    lambda:  |-
      it.rectangle(1, 1, 126, 32);
      it.printf(20, 10, id(five8), "%.1f°C", id(temperature).value);
      it.printf(84, 10, id(five8), "%.0fhPa", id(pressure).value);
      it.printf(5, 26, id(five8), "Temperature Pressure");
      it.strftime(2, 35, id(five8), "Time: %H:%M", id(time).now());
      it.strftime(2, 45, id(five8), "Date: %d.%m.%Y", id(time).now());
      it.printf(2, 60, id(five8), "Wifi: %s", id(status).value ? "Online," : "Offline");
      it.printf(80, 60, id(five8), "%.1fdb", id(signal).value);

```

**Traceback (if applicable):**
```
Compiling .pioenvs/espm3_solar/lib55f/ESP8266WiFi/ESP8266WiFiAP.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:110:37: error: no matching function for call to 'esphomelib::time::SNTPComponent::set_servers(const char [15])'
time->set_servers("0.pool.ntp.org");
^
src/main.cpp:110:37: note: candidate is:
In file included from .piolibdeps/esphomelib/src/esphomelib/application.h:128:0,
from src/main.cpp:2:
.piolibdeps/esphomelib/src/esphomelib/time/sntp_component.h:28:8: note: void esphomelib::time::SNTPComponent::set_servers(const string&, const string&, const string&)
void set_servers(const std::string &server_1,
^
.piolibdeps/esphomelib/src/esphomelib/time/sntp_component.h:28:8: note:   candidate expects 3 arguments, 1 provided
src/main.cpp: In lambda function:
src/main.cpp:117:56: warning: 'esphomelib::sensor::Sensor::<anonymous union>::value' is deprecated (declared at .piolibdeps/esphomelib/src/esphomelib/sensor/sensor.h:129): .value is deprecated. Please use .state instead [-Wdeprecated-declarations]
it.printf(20, 10, five8, "%.1f°C", temperature->value);
^
src/main.cpp:117:56: warning: 'esphomelib::sensor::Sensor::<anonymous union>::value' is deprecated (declared at .piolibdeps/esphomelib/src/esphomelib/sensor/sensor.h:129): .value is deprecated. Please use .state instead [-Wdeprecated-declarations]
src/main.cpp:118:53: warning: 'esphomelib::sensor::Sensor::<anonymous union>::value' is deprecated (declared at .piolibdeps/esphomelib/src/esphomelib/sensor/sensor.h:129): .value is deprecated. Please use .state instead [-Wdeprecated-declarations]
it.printf(84, 10, five8, "%.0fhPa", pressure->value);
^
src/main.cpp:118:53: warning: 'esphomelib::sensor::Sensor::<anonymous union>::value' is deprecated (declared at .piolibdeps/esphomelib/src/esphomelib/sensor/sensor.h:129): .value is deprecated. Please use .state instead [-Wdeprecated-declarations]
src/main.cpp:122:51: warning: 'esphomelib::binary_sensor::BinarySensor::<anonymous union>::value' is deprecated (declared at .piolibdeps/esphomelib/src/esphomelib/binary_sensor/binary_sensor.h:61): .value is deprecated, please use .state instead [-Wdeprecated-declarations]
it.printf(2, 60, five8, "Wifi: %s", status->value ? "Online," : "Offline");
^
src/main.cpp:122:51: warning: 'esphomelib::binary_sensor::BinarySensor::<anonymous union>::value' is deprecated (declared at .piolibdeps/esphomelib/src/esphomelib/binary_sensor/binary_sensor.h:61): .value is deprecated, please use .state instead [-Wdeprecated-declarations]
src/main.cpp:123:50: warning: 'esphomelib::sensor::Sensor::<anonymous union>::value' is deprecated (declared at .piolibdeps/esphomelib/src/esphomelib/sensor/sensor.h:129): .value is deprecated. Please use .state instead [-Wdeprecated-declarations]
it.printf(80, 60, five8, "%.1fdb", signal->value);
^
src/main.cpp:123:50: warning: 'esphomelib::sensor::Sensor::<anonymous union>::value' is deprecated (declared at .piolibdeps/esphomelib/src/esphomelib/sensor/sensor.h:129): .value is deprecated. Please use .state instead [-Wdeprecated-declarations]
Compiling .pioenvs/espm3_solar/lib55f/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
*** [.pioenvs/espm3_solar/src/main.cpp.o] Error 1

```

**Additional information:**

Nope

rradar, 2018-11-17:
When I remove

```
time:
  - platform: sntp
    id: time
    servers:
      - 0.pool.ntp.org
```

and the responding part under display

```
      it.strftime(2, 35, id(five8), "Time: %H:%M", id(time).now());
      it.strftime(2, 45, id(five8), "Date: %d.%m.%Y", id(time).now());
```

I'm able to compile. The problem looks like it's related to the time/sntp feature

roadfox, 2018-11-18:
had the same problem

but as the error message indicates the library expects 3 time servers not only one, so i change to

time:
  - platform: sntp
    id: time
    servers:
      - 0.pool.ntp.org
      - 1.pool.ntp.org
      - 2.pool.ntp.org

and the project compiles again.

in my opinion it does not make much sense that one has to enter 3 ntp servers as the ntp.org addresses return already round robin ip addresses from different dns servers, but hey at least time is accurate ;-)

OttoWinter, 2018-11-18:
Uh yes, I see the error now. I have it fixed locally, though I can't commit it right now as it's entagled with changes from a big refactor to the esphomeyaml code-base.

For working around this until it's fixed, just specify all sntp servers

closed 2018-11-19 by OttoWinter