esphome/esphome #239: Template Cover: MQTT Spamming

ctalkington, 2018-11-21:
**Operating environment (Hass.io/Docker/pip/etc.):**
PIP via Platform IO in VSCODE

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP8266 NODEMCU AMICA

**Affected component:**
https://esphomelib.com/esphomeyaml/components/cover/template.html

**Description of problem:**
I have a Template Cover using lambda for a garage door based on a Template Binary Sensor based on an ultrasonic sensor. The log fills with:

[18:50:08][D][cover.mqtt:070]: 'Garage Door': Sending state open
[18:50:08][D][cover.mqtt:070]: 'Garage Door': Sending state open
[18:50:08][D][cover.mqtt:070]: 'Garage Door': Sending state open
[18:50:08][D][cover.mqtt:070]: 'Garage Door': Sending state open
[18:50:08][D][cover.mqtt:070]: 'Garage Door': Sending state open
[18:50:08][D][cover.mqtt:070]: 'Garage Door': Sending state open

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: ultrasonic
    id: garage_door_ultrasonic
    name: "Garage Door Ultrasonic Sensor"
    trigger_pin: D1
    echo_pin: D2

binary_sensor:
  - platform: template
    id: garage_door_open
    name: "Garage Door Open"
    lambda: >-
      if (isnan(id(garage_door_ultrasonic).state)) {
        return {};
      } else if (id(garage_door_ultrasonic).state > .5) {
        return true;
      } else {
        return false;
      }

cover:
  - platform: template
    name: "Garage Door"
    lambda: >-
      if (id(garage_door_open).state) {
        return cover::COVER_OPEN;
      } else {
        return cover::COVER_CLOSED;
      }
    open_action:
      - switch.turn_off:
          id: garage_door_relay
      - switch.turn_on:
          id: garage_door_relay
      - delay: 0.1s
      - switch.turn_off:
          id: garage_door_relay
    close_action:
      - switch.turn_off:
          id: garage_door_relay
      - switch.turn_on:
          id: garage_door_relay
      - delay: 0.1s
      - switch.turn_off:
          id: garage_door_relay
```

**Traceback (if applicable):**

N/A

**Additional information:**

N/A

postlund, 2018-11-22:
I have the same issue, even when hardcoding the lambda to return for instance `cover::COVER_OPEN`.

ctalkington, 2018-11-27:
thanks for the fix in 1.9.2!

closed 2018-11-27 by ctalkington