esphome/esphome #243: [Suggestion / Feature Request] Handling outputs directly in components

Melkor82, 2018-11-27:
Hello, I've very new to this world and I really love it. Had my first light on yesterday. 
I'm trying to implement a monochromatic light today and
```
output:
- platform: ledc
  id: pin18
  pin: 18

light:
- platform: monochromatic
  name: "striscia"
  id: striscia
  output: pin18
```
that leads to a compile cpp lines like 

```
output::LEDCOutputComponent *pin18 = App.make_ledc_output(18);
::Application::MakeLight _application_makelight_2 = App.make_monochromatic_light("striscia", pin18);
```
after many lights set it is quite difficult to follow wich light activate which output. Wouldn't be more easy to handle outputs directly in lights section? Something like:

```
light:
- platform: monochromatic
  name: "striscia"
  id: striscia
  output: 
  - platform: gpio
    pin: 18
```
then it could be implemented directly as :

`::Application::MakeLight _application_makelight_2 = App.make_monochromatic_light("striscia", App.make_ledc_output(18));`

OttoWinter, 2018-11-27:
The problem with that is that it's very hard to do with the current design of esphomeyaml. The other issue is that despite it being shorter to write, I think it's going to be more difficult to understand for beginners, as in that case the documentation would become more complex.

I think the solution to your problem is quite easy: Just give your components better ids. For example rename your `pin18` output to `striscia_output`.

Melkor82, 2018-11-27:
Yes, i think i will manage names so I can keep better tracks of them. Thanks fot the amazing work!!

closed 2019-01-08 by OttoWinter