esphome/esphome #244: New integration request: RCWL-0516 (doppler radar)

rradar, 2018-11-27:
**What new integration would you wish to have?**
- RCWL-0516

"RCWL-0516 is a doppler radar microwave motion sensor module which can act as an alternative to a PIR motion sensor" 

**If possible, provide a link to an existing library for the integration:**
- https://github.com/mertenats/Open-Home-Automation/tree/master/ha_mqtt_multisensor
- https://github.com/jxmot/ESP8266-RCWL0516
- Tecnically information: https://github.com/jdesbonnet/RCWL-0516

**Product link:**
https://www.aliexpress.com/item/Free-Shipping-Human-body-induction-module-RCWL-0516-microwave-radar-inductive-switch-module-intelligent-sensing-detector/32712700620.html (US $0.49 / piece)

**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I can't use esphomelib for all of my sensors :crying_cat_face:

Melkor82, 2018-11-27:
it looks like it has only one binary output pin, you can already use a simple binary sensor as is stated here 
[https://esphomelib.com/esphomeyaml/cookbook/pir.html](url)

OttoWinter, 2018-11-27:
Yes, as @Melkor82 says this is a simple [gpio binary sensor](https://esphomelib.com/esphomeyaml/components/binary_sensor/gpio.html) so it's already supported.

This is more of a documentation issue. If you want, you can create a cookbook entry (like the one for the PIR sensor).