esphome/esphome #246: Variable/String substitution

thubot, 2018-11-28:
Maybe I'm missing something, but it appears that yaml does not support (bash-style) variable/string substitution, and esphomeyaml doesn't either.

**Is your feature request related to a problem? Please describe.**

I have multiple group of devices with many devices each that are identical to each other, e.g. sensors and switches where sensor_a is identical in hardware and configuration to sensor_b except for the device name.

To ensure a consistent configuration across all devices in a group, whenever I make larger changes to the configuration, I make the changes to one test device and copy its configuration file over the configuration files of the other devices.

As I also name the sensors on a device in a "devicename sensorname" scheme, I have to change the devicename in a lot of places in a file, in a lot of files. This is tedious.

Now that could probably done with some bash wizardry, but the more elegant solution would to define a DEVICENAME variable and just refer to it like "$DEVICENAME Temperature" in strings, for example. This would reduce my work to only changing a single occurence per file.

Additionally, if it were possible to have this substitution work across !include directives, I could just define a skeleton yaml for each device that all include a common device_sensors.yaml.

This would greatly reduce my effort to maintain a uniform configuration across my devices.

**Describe the solution you'd like**
Allow me to define variables that I can refer to in strings and that get substituted in when I compile.

**Additional context**
Example: A config of
```
substitutions:
  - devicename: "sensor_b"
esphomeyaml:
  - name: "$devicename"
sensors:
  - platform: example
    name: "$devicename example"`
```

yields "sensor_b" as devicename and "sensor_b example" as sensor name.

OttoWinter, 2018-11-28:
This is similar to https://github.com/OttoWinter/esphomeyaml/issues/67 which I rejected because YAML and jinja don't work too well together (this would be bad for future esphomeyaml editor plugins)

This suggestion however I like a lot. Additionally it wouldn't be too hard for me to implement (just a simple pre-processing pass before condig validation). It works well with future tooling and syntax highlighters and it is entirely opt-in. Only if you actually use and have definied a substitution, esphomeyaml will replace it.

Barring any possible major implementation details that stop this, you can expect this for 1.10 😀

roadfox, 2018-11-29:
Ansible config management solves jinja2 - yaml pretty well, lots of curly brackets needed but once used to it, its ok.

OttoWinter, 2018-11-29:
@roadfox There's a big difference between how ansible does it and how it would be required for esphomeyaml. Ansible has jinja2 templates per-yaml-value, but esphomeyaml would need to interpret the whole YAML file as one big jinja2 template. At least in all the editors I've tested that breaks syntax highlighting and json-schema validation support.

The main reason for not choosing jinja2 is that I want to have IDE plugins for esphomeyaml in the future. These would provide syntax highlighting and in-editor validation using the JSON schema validators.

OttoWinter, 2019-02-01:
@andreas-bulling FAQ please

brandond, 2019-02-01:
I just have a template file that I run through [j2cli](https://pypi.org/project/j2cli/) when I add devices. Not great for folks that use the dashboard, but I don't really so it's not a problem for me.

```
esphomeyaml:
  name: fixture{{ID}}
  platform: ESP8266
  board: esp01_1m

sensor:
  - platform: wifi_signal
    name: "Fixture {{ID}} WiFi Signal"
    filters:
      - sliding_window_moving_average:
          window_size: 4
          send_every: 4

output:
  - platform: gpio
    id: relay_output
    pin:
      number: 4
      mode: OUTPUT

light:
  - platform: binary
    name: "Fixture {{ID}} Light"
    id: relay0
    output: relay_output

# wifi; mqtt; etc
```

``bash
for ID in {{00..07}; do ID=$ID j2 fixtures.yaml.j2 > fixture${ID}.yaml; esphomeyaml fixture${ID}.yaml run --no-logs --upload-port OTA; done
```

andreas-bulling, 2019-02-01:
Thanks a lot - found it!

andreas-bulling, 2019-02-01:
Another question (off topic, I know, sorry): I have a Sonoff SV running very nicely now with the native HA interface. At the moment I have a binary_sensor and a switch. How can I instead have a light component (as this is what I control with the relay). What configuration is required?

OttoWinter, 2019-02-01:
@andreas-bulling See sonoff s20 or basic examples

brandond, 2019-02-01:
I actually just pasted an example of that above - binary light with a GPIO output controlling a relay.

OttoWinter, 2019-02-01:
Whoops

closed 2019-02-01 by OttoWinter

OttoWinter, 2019-02-01:
@andreas-bulling Please read the automation guide. It's only a slight modification from your previous setup

brandond, 2019-02-01:
I left out the rest of the config ;) I should put them up on github pages or something. Here are the full templates that I use for my most common devices - a shelly v1 with toggle switch, and a plug with button+leds.

https://gist.github.com/brandond/44afc1e344ed8d6b89107ada11737687

andreas-bulling, 2019-02-01:
Thanks a lot, that is super helpful!