esphome/esphome #248: Can't compile it.strftime with lcd display

rradar, 2018-11-29:
**Operating environment (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266/Board/Sonoff):**

ESP8266
Wemos D1 mini

**Affected component:**

https://esphomelib.com/esphomeyaml/components/display/lcd_gpio.html#display-lcd-lambda

**Description of problem:**

Can't compile `it.strftime("It is %H:%M on %d.%m.%Y");`

**Problem-relevant YAML-configuration entries:**
```yaml
display:
  - platform: lcd_pcf8574
    dimensions: 18x2
    address: 0x27
    lambda: |-
      it.strftime("It is %H:%M on %d.%m.%Y");
```

**Traceback (if applicable):**

```
src/main.cpp: In lambda function:
src/main.cpp:31:44: error: no matching function for call to 'esphomelib::display::LCDDisplay::strftime(const char [24])'
it.strftime("It is %H:%M on %d.%m.%Y");
^
src/main.cpp:31:44: note: candidates are:
In file included from .piolibdeps/esphomelib/src/esphomelib/application.h:39:0,
from src/main.cpp:2:
.piolibdeps/esphomelib/src/esphomelib/display/lcd_display.h:47:8: note: void esphomelib::display::LCDDisplay::strftime(uint8_t, uint8_t, const char*, esphomelib::time::EsphomelibTime)
void strftime(uint8_t column, uint8_t row, const char *format, time::EsphomelibTime time) __attribute__ ((format (strftime, 4, 0)));
^
.piolibdeps/esphomelib/src/esphomelib/display/lcd_display.h:47:8: note:   candidate expects 4 arguments, 1 provided
In file included from .piolibdeps/esphomelib/src/esphomelib/application.h:39:0,
from src/main.cpp:2:
.piolibdeps/esphomelib/src/esphomelib/display/lcd_display.h:49:8: note: void esphomelib::display::LCDDisplay::strftime(const char*, esphomelib::time::EsphomelibTime)
void strftime(const char *format, time::EsphomelibTime time) __attribute__ ((format (strftime, 2, 0)));
^
.piolibdeps/esphomelib/src/esphomelib/display/lcd_display.h:49:8: note:   candidate expects 2 arguments, 1 provided
*** [.pioenvs/d1_lcd/src/main.cpp.o] Error 1

```

**Additional information:**

using `it.print(2, 0, "Hello World!");` prints Hello World like expected on the display

roadfox, 2018-12-02:
you need to add the possition where to print and what to print:

> it.strftime(2, 3, "It is %H:%M on %d.%m.%Y", id(time).now());

closed 2018-12-02 by OttoWinter

rradar, 2018-12-02:
Just works! Excellent! :1st_place_medal: