esphome/esphome #255: ESP8266 doesn't restore switch state on boot

definitio, 2018-12-03:
There is no retained message on command topic. Also tried with `arduino_version: dev`, but It's still not working.

Config:
```
esphomeyaml:
  name: lroom_aquarium
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: !secret ssid
  password: !secret wifi_password
  reboot_timeout: 300s

web_server:
  port: 80

mqtt:
  broker: !secret broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  topic_prefix: 'lroom/aquarium'
  birth_message:
    topic: 'lroom/aquarium/state'
    payload: online
  will_message:
    topic: 'lroom/aquarium/state'
    payload: offline
  discovery: true
  reboot_timeout: 300s

logger:
  baud_rate: 0 # disable logging via UART
  level: VERY_VERBOSE

ota:
  password: !secret ota_password

switch:
  - platform: gpio
    name: 'Aquarium light'
    id: 'aq_light'
    state_topic: 'lroom/aquarium/light'
    command_topic: 'lroom/aquarium/light/set'
    pin: 12
```

Log:
```
[14:19:38][I][mqtt.client:115]: MQTT Connected!
[14:19:39][V][mqtt.client:239]: Publish(topic='lroom/aquarium/state' payload='online' retain=1)
[14:19:39][V][mqtt.client:201]: Subscribing to topic='lroom/aquarium/light/set' qos=0...
[14:19:39][V][mqtt.component:067]: 'Aquarium light': Sending discovery...
[14:19:39][V][mqtt.client:239]: Publish(topic='homeassistant/switch/lroom_aquarium/aquarium_light/config' payload='{"name":"Aquarium light","state_topic":"lroom/aquarium/light","command_topic":"lroom/aquarium/light/set","availability_topic":"lroom/aquarium/state","unique_id":"ESPswitchaquarium_light","device":{"identifiers":"5ccf7f3b57c4","name":"lroom_aquarium","sw_version":"esphomelib v1.9.3 Dec  3 2018, 14:18:08","model":"PLATFORMIO_ESP01_1M","manufacturer":"espressif"}}' retain=1)
[14:19:39][V][mqtt.client:239]: Publish(topic='lroom/aquarium/light' payload='OFF' retain=1)
[14:19:39][I][application:089]: setup() finished successfully!
[14:19:39][I][application:094]: You're running esphomelib v1.9.3 compiled on Dec  32018, 14:18:08
[14:19:39][C][mqtt.client:064]:   Log Topic: 'lroom/aquarium/debug'
[14:19:39][C][mqtt.client:067]:   Availability: 'lroom/aquarium/state'
[14:19:39][C][switch.mqtt:048]: MQTT switch 'Aquarium light':
[14:19:39][C][switch.mqtt:055]:   State Topic: 'lroom/aquarium/light'
[14:19:39][C][switch.mqtt:055]:   Command Topic: 'lroom/aquarium/light/set'
[14:19:39][I][application:108]: Running through first loop()
[14:19:39][I][application:125]: First loop finished successfully!
```

OttoWinter, 2018-12-03:
esphomelib restores the state from RTC memory (*), not MQTT. So the command topic is not marked as retained (though you can entirely do so using the appropriate settings in HA.) There are a few reasons for this, most importantly that the command topic does not always hold the "real" state of the switch, for example when automations are used.

Are you seeing that the switch does not restore the state at all on boot? Then please do two things for me

* Verify what version of esphomelib you're using as the issue template asked you to do 😶
* How many switches does that device have? If it's a lot, it might be because the ESP does not have enough RTC memory. Try reducing the amount of switches and see if that works.

(* except for the ESP32; there it's the nvs flash storage.)

definitio, 2018-12-03:
Sorry, I accidentally deleted the issue template. Added missing information. I wrote about mqtt because it theoretically can cause similar problem if 'OFF' message is retained, but not in my case.
* v1.9.3, you can see it in log.
* Tried with only 1 switch and configuration from the first post.

definitio, 2019-01-11:
Tested with 1.10.0 - nothing changed. State is never restored after power loss, but almost every time after software reboot (after ota or restart switch).

OttoWinter, 2019-01-11:
> State is never restored after power loss, but almost every time after software reboot (after ota or restart switch).

Yes, that is to be expected. RTC memory is lost after power loss.

On the ESP32, that data _is_ stored in NVS (non-volatile storage) so it is restored even after power loss, but that's only possible because they have SDKs that automatically detect bad pages and reduce writer operations. On ESP8266, there are no APIs for automatic flash sector cycling - so if I were to add support for storing state across power losses it could quickly wear out your flash, and that's bad.

closed 2019-01-11 by definitio

abmantis, 2019-03-12:
Espurna and Tasmota have this, even on the ESP8266. Can't we have this feature too?

brandond, 2019-03-13:
@abmantis have you seen https://github.com/esphome/feature-requests/issues/73

abmantis, 2019-03-13:
@brandond Oh I haven't seen that! Great. Thanks.