esphome/esphome #257: remote_receiver error: Data is coming in too fast

spattinson, 2018-12-05:
Env: Home Assistant, 433Mhz generic binary window/door sensors. 
Board: Wemod D1 mini
Component: remote_receiver
Description: Code cannot receive codes using rc_switch, get:
[15:26:32][W][remote.receiver:357]: Data is coming in too fast!
This may be significant:
[15:26:18][W][remote.receiver:320]: Remote Receiver Signal starts with a HIGH value. Usually this means you have to invert the signal using 'inverted: True' !
Where is this supposed to be set?
The sample sketch for rc_switch ReceiveDemo_Advanced works fine
As does OpenMQTTGateway

yaml:

esphomeyaml:
  name: rf_bridge
  platform: ESP8266
  board: d1_mini
wifi:
  ssid: 'home'
  password: 'fileybay'

mqtt:
  broker: '192.168.1.122'
  username: 'scott'
  password: 'XXXXXXX'

# Enable logging
logger:

ota:
  password: 'fileybay'

remote_receiver:
  pin: D2
  dump: rc_switch

Log:
[15:26:18][I][logger:057]: Log initialized
[15:26:18][C][ota:465]: There have been 1 suspected unsuccessful boot attempts.
[15:26:18][I][application:053]: Running through setup()...
[15:26:18][C][remote.receiver:296]: Setting up Remote Receiver...
[15:26:18][C][wifi:027]: Setting up WiFi...
[15:26:18][I][wifi:279]: WiFi Connecting to 'home'...
[15:26:18][I][wifi:347]: WiFi connected!
[15:26:18][C][wifi:325]:   SSID: 'home'
[15:26:18][C][wifi:326]:   IP Address: 192.168.1.203
[15:26:18][C][wifi:328]:   BSSID: D8:50:E6:A8:1E:20
[15:26:18][C][wifi:330]:   Hostname: 'rf_bridge'
[15:26:18][C][wifi:332]:   Channel: 3
[15:26:18][C][wifi:333]:   Subnet: 255.255.255.0
[15:26:18][C][wifi:334]:   Gateway: 192.168.1.1
[15:26:18][C][wifi:335]:   DNS1: 192.168.1.1
[15:26:18][C][wifi:336]:   DNS2: 0.0.0.0
[15:26:18][C][ota:133]: Over-The-Air Updates:
[15:26:18][C][ota:134]:   Address: 192.168.1.203:8266
[15:26:18][C][ota:136]:   Using Password.
[15:26:18][C][mqtt.client:020]: Setting up MQTT...
[15:26:18][I][mqtt.client:079]: Connecting to MQTT...
[15:26:18][I][mqtt.client:115]: MQTT Connected!
[15:26:18][I][application:089]: setup() finished successfully!
[15:26:18][I][application:094]: You're running esphomelib v1.9.0 compiled on Dec  5 2018, 13:59:20
[15:26:18][C][wifi:340]: WiFi:
[15:26:18][C][wifi:325]:   SSID: 'home'
[15:26:18][C][wifi:326]:   IP Address: 192.168.1.203
[15:26:18][C][wifi:328]:   BSSID: D8:50:E6:A8:1E:20
[15:26:18][C][wifi:330]:   Hostname: 'rf_bridge'
[15:26:18][C][wifi:332]:   Channel: 3
[15:26:18][C][wifi:333]:   Subnet: 255.255.255.0
[15:26:18][C][wifi:334]:   Gateway: 192.168.1.1
[15:26:18][C][wifi:335]:   DNS1: 192.168.1.1
[... 73 lines cut ...]
Decimal: 7697716 (24Bit) Binary: 011101010111010100110100 Tri-State: F1FFF1FF01F0 PulseLength: 345 microseconds Protocol: 1
Raw data: 10712,390,998,1060,331,1048,339,1044,343,350,1030,1045,343,350,1033,1040,346,348,1035,1039,348,1036,349,1032,354,340,1040,1040,344,348,1035,1038,352,342,1037,350,1034,1038,354,1026,357,338,1043,1031,357,337,1045,344,1038,

The sensor send each trigger about ten times
If I change my yaml to raw/all instead of rc_switch I get this with esphomeyaml

[13:32:43][D][remote.raw:051]: Received Raw: 1578, -470, 251, -1832, 426, -1665,
 366, -402, 285, -414, 170, -475, 184, -421, 269, -417, 265, -549, 269, -1823, 289, -1805, 275, -427, 260, -441, 238, -477, 51, -2043, 231, -1866, 229, -435, 256, -433, 252, -504, 249, -449, 250, -452, 
[13:32:43][D][remote.raw:064]:   245, -464, 245, -1860, 173, -1930, 164, -1942, 145, -478, 245, -464, 250, -524, 238, -468, 240, -475, 237, -469, 234, -1859, 102, -524, 231, -486, 235, -530, 230, -509, 236, -501, 189, -3245, 235, -505, 230, -510, 230, -10000
[13:32:43][D][remote.raw:051]: Received Raw: 184, -667, 245, -1014, 165, -4345, 245, -728, 242, -740, 240, -737, 228, -777, 224, -4455, 233, -817, 212, -5099, 227, -1427, 50, -1312, 226, -987, 181, -1633, 226, -5135, 214, -1179, 222, -1295, 175, -3487, 166, -4627, 145, -4184, 200, 
[13:32:43][D][remote.raw:064]:   -1807, 162, -10000
[13:32:43][W][remote.receiver:357]: Data is coming in too fast!
[13:32:43][D][remote.raw:064]: Received Raw: 96, -4345, 186, -10000
[13:32:43][D][remote.raw:051]: Received Raw: 77, -3667, 41, -91, 43, -380, 70, -504, 228, -613, 60, -537, 121, -716, 47, -523, 46, -427, 158, -455, 23, -284, 73, -101, 193, -103, 69, -122, 49, -168, 121, -28, 57, -457, 222, -77, 112, -509, 95, -72, 182, -41, 51, -169, 129, -428, 99, 
[13:32:44][D][remote.raw:051]:   -67, 159, -271, 94, -85, 172, -27, 128, -108, 50, -50, 145, -155, 158, -200, 97, -58, 44, -101, 161, -95, 57, -394, 457, -176, 502, -85, 178, -88, 1323, -58, 74, -67, 103, -34, 349, -51, 421, -32, 508, -65, 104, -74, 416, -98, 1567, -59, 13, -97, 1433, 
[13:32:44][D][remote.raw:064]:   -17, 10000
[13:32:44][W][remote.receiver:357]: Data is coming in too fast!
[13:32:44][D][remote.raw:064]: Received Raw: -26, 97, -56, 201, -52, 45, -37, 1141, -67, 320, -31, 206, -108, 18, -45, 10000

Things I have tried:
different GPIOs, I started off using D4 GPIO2, because it has an LED and I could see when data arrived, This worked with other code not esphomeyaml. I am currently using D2 GPIO4 - same issue.
swapped over rf receiver module: same issue
tried increasing the buffer, esp8266 seems to crash mostly if you set it greater than 2kb, still crashes sometimes with 2kb, but when it didn't crash I get the same issue.
Disabled  ota in case they caused overhead. No Difference
Thanks in advance. I'm moving from OPenMQTTGateway to esphomeyaml because I wish to standardise, have deployed one project.

OttoWinter, 2018-12-05:
Can you try with

```yaml
remote_receiver:
  pin:
    number: D2
    mode: INPUT_PULLUP
    inverted: True
```

(and in case that doesn't work maybe also with `inverted: False`

Thanks!

OttoWinter, 2018-12-05:
The first line (`41, -310, 160, -51, 64, -151, 326, -158, 28, -61, 278, -30, 124, -157, 328, -124, 404, -82, 231, -264, 27, -70, 92, -28, 149, -253, 542, -47, 53, -437, 163, -187, 24, -219, 101, -130, 30, -52, 198, -193, 115, -40, 58, -312, 22, -337, 173, `) looks like some garbage data - maybe the RF receiver has a high gain setting or it's just noise.

The second one (`-1157, 317, -414, 1067, -410, 1065, -1154, 343, -1171, 352, -411, 1114, -414, 1114, -1182, 357, -418, 1133, -421, 1130, -1198, 373, -1216, 382, -422, 1177, -429, 1175, -423, 1171, -1236, 383, -1254, 395, -436, 1214, -1280, 404, -1292, 418, `) looks a lot more reasonable for remote codes.

I've seen somebody have this issue - and I didn't have a solution back then. I don't have any RF appliances so I never was able to test this integration except for IR codes.

But the fact that you *do* get valid data has me hopeful right now. I think this might be a combination of multiple issues. Ok, next try:

```yaml
remote_receiver:
  pin:
    number: D2  # any pin should work, just an example
    mode: INPUT_PULLUP  # INPUT probably also could work
    # based on your output, it looks like inverted is not necessary
  # limit to only dumping RCSwitch data - I think the raw data might be slowing down the
  # ESP too much - if it still doesn't work try with raw again (the logs help me diagnose this :P)
  dump: rc_switch
```

And additionally, open up the `<NODE_NAME>/src/main.cpp` file and manually remove the `delay(16);` line near the end of the file. In the next esphomelib version that will no longer be necessary, but for now it should increase the update interval.

spattinson, 2018-12-06:
No Change that I could notice. Still the same error with rc_switch as the dump mode. I have also tried changing the "idle" setting 2, 15 and 20, that did not help either. I also changed the clock speed in platformio.ini to 160Mhz because thats what I was using in Arduino  with the rc_switch sample code which works fine:
```
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(115200);
  mySwitch.enableReceive(D4); 
}

void loop() {
  if (mySwitch.available()) {
    output(mySwitch.getReceivedValue(), mySwitch.getReceivedBitlength(), mySwitch.getReceivedDelay(), mySwitch.getReceivedRawdata(),mySwitch.getReceivedProtocol());
    mySwitch.resetAvailable();
  }
}
```
There is usually some noise generated after reset. In the raw dump below I waited a few seconds after the noise finished before triggering my sensor (its a window open sensor with a reed switch) The data from that single trigger is [17:03.26] timestamp to the end of the log.

```
[17:02:50][I][ota:364]: OTA update finished!
[17:02:50][I][helpers:258]: Rebooting safely... Reason: 'ota'
[17:02:59][I][mqtt.client:115]: MQTT Connected!
[17:02:59][I][application:089]: setup() finished successfully!
[17:02:59][I][application:094]: You're running esphomelib v1.9.0 compiled on Dec  6 2018, 17:01:32
[17:02:59][C][wifi:340]: WiFi:
[17:02:59][C][wifi:325]:   SSID: 'home'
[17:02:59][C][wifi:326]:   IP Address: 192.168.1.203
[17:02:59][C][wifi:328]:   BSSID: D8:50:E6:A8:1E:20
[17:02:59][C][wifi:330]:   Hostname: 'rf_bridge'
[17:02:59][C][wifi:332]:   Channel: 3
[17:02:59][C][wifi:333]:   Subnet: 255.255.255.0
[17:02:59][C][wifi:334]:   Gateway: 192.168.1.1
[17:02:59][C][wifi:335]:   DNS1: 192.168.1.1
[17:02:59][C][wifi:336]:   DNS2: 0.0.0.0
[17:02:59][C][remote.receiver:316]: Remote Receiver:
[17:02:59][C][remote.receiver:317]:   Pin: GPIO2 (Mode: INPUT_PULLUP)
[17:02:59][W][remote.receiver:320]: Remote Receiver Signal starts with a HIGH value. Usually this means you have to invert the signal using 'inverted: True' !
[17:02:59][C][remote.receiver:322]:   Buffer Size: 1000
[17:02:59][C][remote.receiver:323]:   Tolerance: 25%
[17:02:59][C][remote.receiver:324]:   Filter out pulses shorter than: 200 us
[17:02:59][C][remote.receiver:325]:   Signal is done after 10000 us of no changes
[17:02:59][C][ota:133]: Over-The-Air Updates:
[17:02:59][C][ota:134]:   Address: 192.168.1.203:8266
[17:02:59][C][ota:136]:   Using Password.
[17:02:59][C][mqtt.client:054]: MQTT:
[17:02:59][C][mqtt.client:055]:   Server Address: 192.168.1.122:1883
[17:02:59][C][mqtt.client:056]:   Username: 'scott'
[17:02:59][C][mqtt.client:057]:   Client ID: 'rf_bridge-dc4f221d8d57'
[17:02:59][C][mqtt.client:059]:   Discovery prefix: 'homeassistant'
[17:02:59][C][mqtt.client:060]:   Discovery retain: YES
[17:02:59][C][mqtt.client:062]:   Topic Prefix: 'rf_bridge'
[17:02:59][C][mqtt.client:064]:   Log Topic: 'rf_bridge/debug'
[17:02:59][C][mqtt.client:067]:   Availability: 'rf_bridge/status'
[17:02:59][I][application:108]: Running through first loop()
[17:02:59][I][application:125]: First loop finished successfully!
[17:02:59][D][remote.raw:051]: Received Raw: 309, -285, 257, -604, 318, -426, 208, -409, 365, -222, 382, -310, 307, -340, 254, -284, 291, -298, 600, -387, 439, -386, 563, -1034, 558, -234, 315, -411, 278, -232, 613, -380, 433, -451, 340, -328, 270, -663, 505, -523, 503, -215, 217, 
[17:02:59][D][remote.raw:051]:   -361, 467, -367, 764, -404, 492, -265, 399, -283, 314, -512, 267, -270, 433, -318, 599, -353, 621, -451, 517, -472, 898, -212, 274, -709, 277, -895, 595, -299, 389, -245, 298, -1027, 373, -358, 445, -448, 345, -341, 355, -244, 464, -217, 267, -202, 
[17:02:59][D][remote.raw:051]:   896, -558, 245, -494, 539, -290, 492, -328, 248, -316, 1305, -648, 297, -780, 491, -368, 662, -285, 256, -345, 832, -662, 259, -444, 355, -395, 468, -387, 469, -1011, 576, -835, 223, -496, 627, -768, 490, -379, 425, -282, 341, -381, 400, -273, 237, 
[17:02:59][D][remote.raw:051]:   -360, 665, -979, 614, -207, 294, -510, 400, -1065, 299, -510, 606, -242, 564, -625, 427, -440, 611, -423, 655, -610, 355, -389, 617, -581, 351, -271, 311, -285, 388, -767, 897, -388, 759, -520, 206, -248, 1102, -608, 353, -301, 233, -214, 797, -407, 
[17:02:59][D][remote.raw:051]:   283, -316, 252, -505, 1332, -436, 202, -354, 208, -205, 532, -440, 773, -631, 368, -295, 498, -473, 688, -210, 664, -247, 269, -305, 493, -274, 215, -305, 204, -512, 227, -495, 446, -254, 257, -499, 584, -274, 263, -535, 212, -594, 214, -323, 498, 
[17:02:59][D][remote.raw:051]:   -232, 798, -303, 374, -348, 322, -685, 254, -282, 309, -433, 615, -419, 781, -221, 319, -240, 477, -322, 224, -508, 230, -343, 747, -309, 201, -472, 626, -536, 361, -458, 331, -739, 234, -304, 317, -354, 212, -296, 328, -250, 388, -477, 344, -226, 381, 
[17:02:59][D][remote.raw:051]:   -693, 392, -326, 575, -548, 230, -324, 204, -597, 811, -794, 1009, -239, 217, -676, 562, -623, 360, -740, 1181, -204, 244, -343, 679, -394, 228, -353, 335, -375, 283, -230, 282, -535, 510, -296, 655, -387, 797, -607, 257, -398, 540, -335, 221, -311, 
[17:02:59][D][remote.raw:051]:   593, -402, 309, -230, 259, -338, 358, -454, 346, -315, 614, -221, 750, -928, 377, -385, 584, -316, 524, -234, 431, -550, 209, -525, 990, -428, 374, -206, 312, -219, 470, -484, 306, -903, 434, -936, 359, -253, 761, -815, 252, -347, 713, -376, 299, -333, 
[... cut ...]
[17:03:26][D][remote.raw:064]:   385, -1058, 1110, -348, 379, -1061, 382, -1054, 385, -10000
[17:03:26][W][remote.receiver:357]: Data is coming in too fast!
[17:03:26][D][remote.raw:051]: Received Raw: 1137, -312, 407, -1036, 398, -1040, 1121, -326, 1123, -334, 391, -1048, 397, -1043, 932, -523, 388, -1047, 397, -1049, 1109, -332, 1122, -334, 388, -1056, 392, -1048, 392, -1050, 1113, -331, 1120, -339, 387, -1061, 1110, -344, 1115, -346, 
[17:03:26][D][remote.raw:064]:   385, -1064, 1113, -348, 381, -1062, 382, -1062, 387, -10000
```

spattinson, 2018-12-07:
I am now getting codes and they look consistent. I receive data is coming in too fast still several times per second constantly.
```
[10:10:22][I][mqtt.client:209]: MQTT Connected!
[10:10:22][I][application:089]: setup() finished successfully!
[10:10:22][I][application:094]: You're running esphomelib v1.10.0-dev compiled on Dec  7 2018, 10:09:06
[10:10:22][C][wifi:338]: WiFi:
[10:10:22][C][wifi:323]:   SSID: 'home'
[10:10:22][C][wifi:324]:   IP Address: 192.168.1.203
[10:10:22][C][wifi:326]:   BSSID: D8:50:E6:A8:1E:20
[10:10:22][C][wifi:328]:   Hostname: 'rf_bridge'
[10:10:22][C][wifi:330]:   Channel: 3
[10:10:22][C][wifi:331]:   Subnet: 255.255.255.0
[10:10:22][C][wifi:332]:   Gateway: 192.168.1.1
[10:10:22][C][wifi:333]:   DNS1: 192.168.1.1
[10:10:22][C][wifi:334]:   DNS2: 0.0.0.0
[10:10:22][C][logger:094]: Logger:
[10:10:22][C][logger:095]:   Level: DEBUG
[10:10:22][C][logger:096]:   Log Baud Rate: 115200
[10:10:22][C][remote.receiver:312]: Remote Receiver:
[10:10:22][C][remote.receiver:313]:   Pin: GPIO4 (Mode: INPUT_PULLUP)
[10:10:22][W][remote.receiver:316]: Remote Receiver Signal starts with a HIGH value. Usually this means you have to invert the signal using 'inverted: True' !
[10:10:22][C][remote.receiver:318]:   Buffer Size: 1000
[10:10:22][C][remote.receiver:319]:   Tolerance: 25%
[10:10:22][C][remote.receiver:320]:   Filter out pulses shorter than: 200 us
[10:10:22][C][remote.receiver:321]:   Signal is done after 10000 us of no changes
[10:10:22][C][ota:140]: Over-The-Air Updates:
[10:10:22][C][ota:141]:   Address: 192.168.1.203:8266
[10:10:22][C][ota:143]:   Using Password.
[10:10:22][W][ota:147]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts
[10:10:22][C][mqtt.client:058]: MQTT:
[10:10:22][C][mqtt.client:060]:   Server Address: 192.168.1.122:1883 (192.168.1.122)
[10:10:22][C][mqtt.client:061]:   Username: 'scott'
[10:10:22][C][mqtt.client:062]:   Client ID: 'rf_bridge-dc4f221d8d57'
[10:10:22][C][mqtt.client:064]:   Discovery prefix: 'homeassistant'
[10:10:22][C][mqtt.client:065]:   Discovery retain: YES
[10:10:22][C][mqtt.client:067]:   Topic Prefix: 'rf_bridge'
[10:10:23][C][mqtt.client:069]:   Log Topic: 'rf_bridge/debug'
[10:10:23][C][mqtt.client:072]:   Availability: 'rf_bridge/status'
[10:10:23][I][application:108]: Running through first loop()
[10:10:23][I][application:135]: First loop finished successfully!
[10:10:23][W][remote.receiver:328]: Data is coming in too fast!
[10:10:23][W][remote.receiver:328]: Data is coming in too fast!
[10:10:23][W][remote.receiver:328]: Data is coming in too fast!
[10:10:24][W][remote.receiver:328]: Data is coming in too fast!
[10:10:24][W][remote.receiver:328]: Data is coming in too fast!
[10:10:24][W][remote.receiver:328]: Data is coming in too fast!
[10:10:25][W][remote.receiver:328]: Data is coming in too fast!
[10:10:25][W][remote.receiver:328]: Data is coming in too fast!
[10:10:25][W][remote.receiver:328]: Data is coming in too fast!
[10:10:25][W][remote.receiver:328]: Data is coming in too fast!
[10:10:25][W][remote.receiver:328]: Data is coming in too fast!
[10:10:26][W][remote.receiver:328]: Data is coming in too fast!
[10:10:26][W][remote.receiver:328]: Data is coming in too fast!
[10:10:26][W][remote.receiver:328]: Data is coming in too fast!
[10:10:26][W][remote.receiver:328]: Data is coming in too fast!
[10:10:27][W][remote.receiver:328]: Data is coming in too fast!
[10:10:27][W][remote.receiver:328]: Data is coming in too fast!
[10:10:27][W][remote.receiver:328]: Data is coming in too fast!
[10:10:28][W][remote.receiver:328]: Data is coming in too fast!
[10:10:28][W][remote.receiver:328]: Data is coming in too fast!
[10:10:28][W][remote.receiver:328]: Data is coming in too fast!
[10:10:28][W][remote.receiver:328]: Data is coming in too fast!
[10:10:29][W][remote.receiver:328]: Data is coming in too fast!
[10:10:29][W][remote.receiver:328]: Data is coming in too fast!
[10:10:29][W][remote.receiver:328]: Data is coming in too fast!
[10:10:29][W][remote.receiver:328]: Data is coming in too fast!
[10:10:30][W][remote.receiver:328]: Data is coming in too fast!
[10:10:30][D][remote.rc_switch:236]: Received RCSwitch: protocol=1 data='100110010011000110110100'
[10:10:30][D][remote.rc_switch:236]: Received RCSwitch: protocol=4 data='100110010011000110110100'
[10:10:30][D][remote.rc_switch:236]: Received RCSwitch: protocol=1 data='100110010011000110110100'
[... 210 lines cut ...]
rf_bridge/debug [D][binary_sensor:027]: 'Window bathroom': Sending state OFF
rf_bridge/binary_sensor/window_bathroom/state OFF
rf_bridge/debug [D][binary_sensor:027]: 'Window bathroom': Sending state ON
rf_bridge/binary_sensor/window_bathroom/state ON
rf_bridge/debug [D][binary_sensor:027]: 'Window bathroom': Sending state OFF
rf_bridge/binary_sensor/window_bathroom/state OFF
rf_bridge/debug [D][binary_sensor:027]: 'Window bathroom': Sending state ON
rf_bridge/binary_sensor/window_bathroom/state ON
rf_bridge/debug [D][binary_sensor:027]: 'Window bathroom': Sending state OFF
rf_bridge/binary_sensor/window_bathroom/state OFF
rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=4 data='100110010011'
rf_bridge/debug [W][remote.receiver:328]: Data is coming in too fast!
rf_bridge/debug [W][remote.receiver:328]: Data is coming in too fast!
rf_bridge/debug [W][remote.receiver:328]: Data is coming in too fast!
rf_bridge/debug [W][remote.receiver:328]: Data is coming in too fast!
[... cut ...]
rf_bridge/debug [W][remote.receiver:328]: Data is coming in too fast!
rf_bridge/debug [W][remote.receiver:328]: Data is coming in too fast!
rf_bridge/debug [W][remote.receiver:328]: Data is coming in too fast!
hw/debug [D][sensor.dallas:157]: 'Hot Water Temperature': Got Temperature=56.8°C
rf_bridge/debug [W][remote.receiver:328]: Data is coming in too fast!
rf_bridge/debug [W][remote.receiver:328]: Data is coming in too fast!
rf_bridge/debug [W][remote.receiver:328]: Data is coming in too fast!
```

OttoWinter, 2018-12-07:
That's great! I wonder if I should increase the default remote receiver buffer size for the ESP8266. The ESP32 already has 10kb of buffer allocated by the default (ESP8266 uses 1kb by default). Maybe increasing it would help.

You can filter out the duplicate pulses using binary sensor filters:

```yaml
remote_receiver:
  pin: D2
  dump: rc_switch
  filter: 200us
  buffer_size: 10kb

binary_sensor:
  - platform: remote_receiver
    name: "Window bathroom"
    rc_switch_raw:
      code: '100110010011000110110100'
      protocol: 1
    filters:
      - delayed_off: 1s
```

spattinson, 2018-12-07:
Increasing the buffer size causes crashes, this is with 4kb, even with 2kb I get crashes. Why would it be trying to alloc 31736 bytes? device only has 80kb total for user data?

```
[15:36:59]
[15:36:59]Exception (29):
WARNING Exception type: unknown
[15:36:59]epc1=0x402099da epc2=0x00000000 epc3=0x00000000 excvaddr=0x00003dfc depc=0x00000000
WARNING Decoded 0x402099da: void std::vector<int, std::allocator<int> >::_M_emplace_back_aux<int>(int&&)
[15:36:59]
[15:36:59]ctx: cont 
[15:36:59]sp: 3ffffd00 end: 3fffffd0 offset: 01a0
[15:36:59]
[15:36:59]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[15:36:59]3ffffea0:  00000000 00000000 00000000 fffffffe  
[15:36:59]3ffffeb0:  ffffffff 3fffc6fc 00000001 3fff0c28  
[15:36:59]3ffffec0:  00000000 3fff27c4 3fff0eac 3fffff30  
[15:36:59]3ffffed0:  00000000 4020bd38 00007bf8 401004e8  
WARNING Decoded 0x4020bd38: operator new(unsigned int)
WARNING Decoded 0x401004e8: malloc
[15:36:59]3ffffee0:  00000000 402099c7 00007bf8 4020bd38  
WARNING Decoded 0x402099c7: void std::vector<int, std::allocator<int> >::_M_emplace_back_aux<int>(int&&)
WARNING Decoded 0x4020bd38: operator new(unsigned int)
[15:36:59]3ffffef0:  3fff0fe8 00001efe 3fff2594 402099c7  
WARNING Decoded 0x402099c7: void std::vector<int, std::allocator<int> >::_M_emplace_back_aux<int>(int&&)
[15:36:59]3fffff00:  3fff2594 3fff253c 0000008d ffffd8f0  
[15:36:59]3fffff10:  3fff2594 3fff253c 0000006c 40209a20  
WARNING Decoded 0x40209a20: void std::vector<int, std::allocator<int> >::emplace_back<int>(int&&)
[15:36:59]3fffff20:  00000000 3fff253c 0000008d 40209b65  
WARNING Decoded 0x40209b65: esphomelib::remote::RemoteReceiverComponent::loop()
[15:36:59]3fffff30:  ffffd8f0 00000000 4bc6a7f0 00000000  
[15:36:59]3fffff40:  0000006c 00003dfc 00002710 a10624dd  
[15:36:59]3fffff50:  00000000 00000000 3fff1f2c 3fff253c  
[15:36:59]3fffff60:  3fff27d4 00000002 3fff253c 40204d29  
WARNING Decoded 0x40204d29: esphomelib::Component::loop_()
[15:36:59]3fffff70:  3fff27e8 00000002 3fff0c28 40202dd0  
WARNING Decoded 0x40202dd0: esphomelib::Application::loop()
[15:36:59]3fffff80:  00000000 3fff27e8 00000018 3fff25ec  
[15:36:59]3fffff90:  3fffff70 feefeffe feefeffe 3fff0eac  
[15:36:59]3fffffa0:  3fffdad0 00000000 3fff0ea4 402027ec  
WARNING Decoded 0x402027ec: loop
[15:36:59]3fffffb0:  3fffdad0 00000000 3fff0ea4 4020bf78  
WARNING Decoded 0x4020bf78: loop_wrapper() at core_esp8266_main.cpp
[15:36:59]3fffffc0:  feefeffe feefeffe 3ffe8648 40100721  
WARNING Decoded 0x40100721: cont_wrapper
[15:36:59]<<<stack<<<
[15:36:59]
[15:36:59]last failed alloc call: 402099C7(31736)
WARNING Memory allocation of 31736 bytes failed at 402099C7
WARNING Decoded 0x402099c7: void std::vector<int, std::allocator<int> >::_M_emplace_back_aux<int>(int&&)
[15:36:59]
[15:36:59] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[15:36:59]
[15:36:59]load 0x4010f000, len 1384, room 16 
[15:36:59]tail 8
[15:36:59]chksum 0x2d
[15:36:59]csum 0x2d
[15:36:59]vbb28d4a3
[15:36:59]~ld
```
The message about data coming in too quickly is too regular to be actual noise/data I think? Here some log with dump to raw. The message about data coming in too quickly only happens when there is no data and it is every 0.8 seconds approximately:

```
16:02:17][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:18][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:18][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:19][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:20][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:21][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:22][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[... 84 lines cut ...]
[16:02:36][D][remote.raw:064]:   1039, -355, 338, -1026, 1060, -331, 357, -1025, 363, -1216, 10000
[16:02:37][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:38][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:39][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:39][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:40][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:41][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:42][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:42][D][remote.raw:051]: Received Raw: -440, 528, -449, 1209, -972, 268, -624, 438, -264, 657, -370, 214, -338, 611, -296, 965, -585, 482, -375, 1061, -237, 435, -357, 381, -1314, 456, -335, 341, -284, 366, -1261, 612, -477, 205, -615, 320, -314, 409, -211, 227, -270, 372, -688, 
[16:02:42][D][remote.raw:051]:   356, -227, 231, -219, 220, -270, 204, -629, 408, -765, 408, -283, 347, -228, 205, -379, 273, -420, 495, -472, 327, -319, 352, -212, 451, -299, 425, -401, 531, -282, 340, -455, 607, -299, 378, -271, 430, -242, 441, -884, 223, -457, 554, -482, 259, -279, 
[16:02:42][D][remote.raw:051]:   543, -906, 441, -312, 833, -378, 597, -597, 367, -362, 315, -372, 1076, -978, 393, -586, 426, -1370, 384, -5290, 1592, -220, 235, -272, 253, -1417, 507, -362, 437, -231, 268, -373 [... cut ...]
[16:02:46][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:47][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:47][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:48][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:49][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[16:02:50][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
```

OttoWinter, 2018-12-19:
Looked into this again and the `10kb` is a bit wrong. It's allocating a buffer with 10 000 integers of 4 byte each. So actually it's 40kb - which is too much. Maybe try other numbers, `5kb` or `3kb`

Looking at RCSwitch again, I guess some of the computation could be lifted into the interrupt, but I don't really like that solution. 

I find it really weird though that somehow there can be that much data in a short amount of time. I'm confused

spattinson, 2018-12-28:
I have tried other buffer sizes 2kb works sometimes but I get crashes. Stack trace is in this  https://github.com/OttoWinter/esphomeyaml/issues/257#issuecomment-445279239
I have had this deployed for a few weeks and it is mostly working despite the "data coming in too quickly" messages. I have had to reset the module a few times as it stopped updating my hass when motion sensors are triggered. I dont know what the cause is so I will leave my computer on collecting the debug log.

There are two sensors out of ten which do not work. Guessing the timing may be slightly off from what the code is expecting.  One produces inconsistant codes, these are the same sensor being triggered and I have removed the lines in between, note detected as protocol 1 and 4, the beginning of the code is truncated by a varying amount

```rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=1 data='10011000100110011'
rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=4 data='10011000100110011'
rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=1 data='00100110011'
rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=4 data='00100110011'
rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=1 data='1110011000100110011'
rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=4 data='1110011000100110011'
rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=1 data='11110011000100110011'
rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=4 data='11110011000100110011'
rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=1 data='0011000100110011'
rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=4 data='0011000100110011'
rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=1 data='1110011000100110011'
rf_bridge/debug [D][remote.rc_switch:236]: Received RCSwitch: protocol=4 data='1110011000100110011'
```
The other sensor which does not work is not detected as rc_switch protocol at all, have set "all" as the protocol:

```[17:01:49][D][remote.raw:051]: Received Raw: 946, -528, 215, -296, 202, -322, 763, -442, 369, -752, 328, -622, 670, -207, 227, -365, 574, -243, 1416, -545, 499, -451, 573, -424, 1191, -228, 723, -342, 226, -201, 681, -299, 437, -431, 548, -226, 252, -206, 611, -697, 519, -351, 251, 
[17:01:49][D][remote.raw:051]:   -263, 228, -355, 223, -399, 674, -885, 329, -453, 559, -760, 307, -277, 1048, -1368, 252, -399, 359, -330, 384, -796, 564, -770, 493, -455, 314, -439, 346, -827, 429, -350, 228, -308, 379, -318, 243, -240, 318, -207, 209, -464, 248, -588, 731, -377, 
[17:01:49][D][remote.raw:051]:   340, -636, 501, -529, 310, -327, 210, -349, 2280, -3546, 284, -4473, 210, -267, 1030, -506, 534, -338, 663, -1286, 455, -598, 1240, -2589, 243, -581, 1141, -1197, 2576, -408, 363, -224, 307, -264, 1328, -492, 330, -261, 362, -536, 581, -599, 944, -431, 
[17:01:49][D][remote.raw:051]:   515, -993, 394, -324, 330, -334, 767, -206, 313, -471, 399, -447, 229, -223, 713, -401, 301, -405, 592, -334, 1083, -415, 844, -203, 309, -325, 1023, -419, 680, -434, 557, -432, 297, -408, 404, -221, 375, -285, 449, -924, 542, -608, 301, -495, 392, 
[17:01:49][D][remote.raw:051]:   -445, 398, -357, 361, -325, 958, -557, 284, -531, 245, -2323, 1707, -236, 695, -794, 286, -369, 294, -932, 475, -345, 403, -309, 341, -873, 434, -508, 475, -466, 258, -710, 244, -234, 285, -1182, 403, -306, 209, -592, 593, -421, 356, -475, 777, -412, 
[17:01:49][D][remote.raw:064]:   261, -412, 2224, -523, 1818, -579, 615, -1746, 1798, -605, 618, -1815, 613, -1806, 615, -10000
[17:01:50][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
[17:01:50][D][remote.raw:051]: Received Raw: 734, -333, 929, -465, 603, -360, 281, -1293, 206, -260, 248, -248, 823, -363, 416, -896, 458, -233, 270, -681, 345, -980, 230, -415, 609, -513, 345, -381, 206, -252, 353, -379, 208, -558, 857, -505, 223, -1505, 389, -1032, 524, -543, 212, 
[17:01:50][D][remote.raw:064]:   -2058, 1670, -487, 551, -1505, 517, -1452, 1459, -479, 1444, -481, 475, -1411, 1400, -474, 1391, -476, 455, -1390, 1368, -470, 451, -1376, 450, -1366, 446, -10000
[17:01:50][D][remote.raw:064]: Received Raw: 550, -1560, 577, -1668, 626, -10000
[17:01:50][D][remote.raw:064]: Received Raw: 10000
[17:01:50][D][remote.raw:051]: Received Raw: -597, 1037, -2180, 2589, -507, 431, -520, 1598, -540, 1703, -1459, 623, -298, 418, -424, 327, -4769, 225, -601, 213, -243, 371, -426, 557, -488, 556, -363, 379, -335, 406, -1085, 1335, -1499, 1567, -414, 824, -2114, 1556, -407, 1527, -416, 
[17:01:50][D][remote.raw:064]:   547, -1355, 1470, -408, 1459, -414, 519, -1324, 1421, -409, 1418, -411, 1417, -414, 498, -1311, 492, -1298, 1388, -409, 1391, -420, 480, -1315, 1368, -435, 1360, -465, 431, -1369, 1322, -483, 423, -1377, 416, -1364, 424, -10000
[17:01:50][D][remote.raw:051]: Received Raw: 471, -1286, 485, -1289, 483, -1298, 1355, -440, 1346, -454, 1348, -449, 450, -1331, 1345, -450, 1350, -458, 440, -1339, 1233, -549, 1348, -449, 1347, -461, 440, -1340, 449, -1333, 1340, -449, 1344, -457, 436, -1343, 1342, -447, 1346, -454, 
[17:01:50][D][remote.raw:064]:   441, -1340, 1359, -450, 239, -1537, 434, -1329, 447, -10000
[... cut ...]  -383, 735, 
[17:01:52][D][remote.raw:051]:   -633, 327, -472, 367, -224, 205, -522, 271, -811, 1001, -256, 506, -361, 276, -345, 955, -292, 459, -772, 231, -462, 514, -589, 516, -274, 263, -461, 270, -669, 862, -358, 775, -478, 684, -366, 404, -268, 355, -522, 319, -498, 304, -316, 248, -581, 
[17:01:52][D][remote.raw:064]:   279, -237, 292, -997, 242, -1109, 426, -4169, 5501, -2956, 1113, -10000
[17:01:52][W][remote.receiver:328]: Data is coming in too fast! Try increasing the buffer size.
```