esphome/esphome #259: SI7021 Heater

ThellraAK, 2018-12-06:
**Is your feature request related to a problem? Please describe.**

I live in southeast Alaska which is a temperate rainforest, as such we often have condensing humidity.

The SI701's humidity sensor is damaged when it is above 80% RH, as well as when it has condensing humidity.

**Describe the solution you'd like**

Ability to turn on SI7021's built in heater, preferably with the ability to give it corrected temperature and pressure in order to get an accurate RH%

**Additional context**
What I have done for another sensor, is have a SI7021 and a BMP280 which given pressure and ambient temperature, you can get corrected humidity, this is written for arduino and I can share the code to do it.

I'm not sure if it would be within the scope of esphomeyaml, but it would be nice to be able to both enable the heater, and be able to provide it with a temperature and pressure so accurate RH% can be given

ThellraAK, 2019-01-04:
@OttoWinter I could have sworn I put the code I used here from ubuntu pastebin.

Would you be interested in how I have done this on a itsbitsym0 (Arduino zero type clone)

OttoWinter, 2019-01-08:
@ThellraAK Sure, you can post it here. As you might see from the amount of feature requests though, it's impossible to implement everything. I would of course accept a PR though that implements this as an option.