esphome/esphome #261: Override component type used for mqtt auto discovery

oscar-b, 2018-12-07:
**Is your feature request related to a problem? Please describe.**
Adding something like a Shelly to HA includes adding the relay as:

```
switch:
  - platform: gpio
```

This means that esphomeyaml will announce it as a switch to HA via mqtt auto discovery. Often though, a relay controls lights, which would make more sense as an entity in HA.

**Describe the solution you'd like**
Being able to override the component type (in this case 'switch') being announced to HA, perhaps with an option to the `switch` base domain:
https://github.com/OttoWinter/esphomeyaml/blob/7c7032c59e4afb548747e1fd62d9b9871f130020/esphomeyaml/components/switch/__init__.py#L139

brandond, 2018-12-08:
you can set the `icon` property on the switch entry to change how it shows up in HA; does that do what you're looking for?

```yaml
switch:
  - platform: gpio
    name: "Floor Lamp"
    icon: "mdi:lamp"
    pin: 13
```

escoand, 2018-12-08:
I solved the save problem in that way:

```yaml
light:
  - platform: binary
    name: Licht
    id: light
    output: relay
output:
  - platform: gpio
    pin: GPIO5
    id: relay
```

OttoWinter, 2018-12-08:
In esphomelib, as in HA, a `switch` is a switch and a `light` is a light. They are very different architecturally so you cannot simply create a light out of a switch because they are fundamentally different.

There are two options for you:

 * Make it a light in esphomelib, like @escoand suggested.
 * Make it a light on the HA side, for example using the [`light.switch`](https://www.home-assistant.io/components/light.switch/) platform.

[image]

closed 2018-12-08 by OttoWinter

oscar-b, 2018-12-08:
@OttoWinter Thanks, I ended up with something like this, would you be interested to have this added as an example in the cookbook section? Should be the same for the Sonoff I guess.

```yaml
esphomeyaml:
  name: kitchen_window_shelly1
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

light:
  - platform: binary
    id: light
    name: "Kitchen lights"
    output: relay

output:
  - platform: gpio
    id: relay
    pin: GPIO4

binary_sensor:
  - platform: gpio
    id: sw
    pin: GPIO5
    on_press:
      then:
        - light.toggle: light
    on_release:
      then:
        - light.toggle: light
```

OttoWinter, 2018-12-11:
@oscar-b Yes.