esphome/esphome #262: Pulse counter as input on ESP32

voicevon, 2018-12-08:
```
esphomeyaml:
  name: ver30
  platform: ESP32
  board: nodemcu-32s
sensor:
  - platform: pulse_counter
    name: "flow"
    id: flow
    filters: []
    pin: 
      number: 39  #  18
    #   mode: INPUT_PULLUP
```
The above yaml can't be compiled with ` - Invalid config for [sensor.pulse_counter]: ESP32: Pin 39 (34-39) can only be used as input pins. for dictionary value @ data['pin']['number']. Got 39. (See /config/esphomeyaml/Ver30.yaml, line 112). `
But I modified it in main.cpp, and compiled it in vsCode, then I uploaded to esp32, the pulse counter works well. That proved pulse counter can work as an INPUT_PIN.

voicevon, 2018-12-09:
I tried to make GPIO 35 as gpio_binary_sensor(with external pull-up resistor 20k),  Yaml is
```
  - platform: gpio
    name: "sp"
    id: sp
    on_press:
        then:
          - script.execute: my_script
    on_release:
        then:
          - script.execute: my_script 
    device_class: lock
    pin: 
      number: 35
    #   mode: INPUT_PULLUP
      inverted: true
```
When I compile this in Hassio/Esphomeyaml/Compile, I got info as:
```
Failed config
 binary_sensor.gpio: 
   - Invalid config for [binary_sensor.gpio]: ESP32: Pin 35 (34-39) can only be used as input pins. for dictionary value @ data['pin']['number']. Got 35. (See /config/esphomeyaml/Ver30.yaml, line 63). 
    - platform: gpio
      device_class: lock
      id: sp
      name: sp
      on_press: [source /config/esphomeyaml/Ver30.yaml:67]
        then: ?
          - script.execute: my_script
      on_release: [source /config/esphomeyaml/Ver30.yaml:70]
        then: ?
          - script.execute: my_script
      pin: [source /config/esphomeyaml/Ver30.yaml:74]
        inverted: True
        number: 35
```
Then I compiled in esphomelib. Everything is working well. the MQTT dash-board showed the sensor state correctly.

OttoWinter, 2018-12-11:
Fixes in https://github.com/OttoWinter/esphomeyaml/pull/234 - was a small typo in validation code.

closed 2018-12-11 by OttoWinter