esphome/esphome #263: esphomeyaml dashboard - upload to IP address

ladefoged81, 2018-12-08:
When using the command line its possible to use --upload-port x.x.x.x to overcome DNS issues when uploading.
This feature could be nice to have in the dashboard.

OttoWinter, 2018-12-11:
Two reason why this doesn't exist:

- a) I don't want to add a text field in the dashboard interface (read as: I don't know how to make the upload `<select>` also work as a text field).
- b) I don't understand why you would not fully switch to static IPs if you're experiencing issues with mDNS. Each time you do an OTA, you will have to look up the current static IP manually and then type that in again and again. Plus with static IPs you get all other benefits too, like for example much faster connection times because of no DHCP.

ladefoged81, 2018-12-11:
I only have one esp module that I have to use --upload-port with and it's only the docker container that can't find it for some reason. It works fine with the other 8 esp modules I currently have.
I can ping the esp module from everything else using the dns name without issues.

OttoWinter, 2018-12-19:
Let's not fight symptoms, but the root issue.

Can you try running the docker container with the `--net=host` option and see if that helps? The Hass.io add-on will switch to that networking mode in the next version and that seems to fix all the mDNS troubles.

ladefoged81, 2018-12-19:
Yes I use host network and it does not solve the issue.
I have a hunch (not verified) it's my Ubiquiti setup that does something wierd, but I have no idea why. The setup is pretty standard.

RaveGun, 2018-12-19:
Not that I have any idea how this mDNS works but on my setup OTA does not work either. Is maybe a possibility to add some ota: configuration item, like fallback_IP in case the dslookup does not work?

OttoWinter, 2019-01-08:
@ladefoged81 @RaveGun I mean it's all possible, but I want to keep the configuration as clean as possible. And by the way, what speaks against just giving the ESP a static IP (like the error message suggests)? If you're not using static IP addresses, a `fallback_ip` option or similar will never be accurate, because the ESP might have received a new DHCP lease.

RaveGun, 2019-01-16:
@OttoWinter 
I am sorry to answer this so late.

My nodes are somehow production nodes and once configured and up and running I try not to change them just maybe add more nodes. There are some development nodes that I update just so but are not vital.

If a plan on updating a node I will, for sure, have edit the configuration yaml. So I could search the IP and change the fallback value. I have only an couple of fixed ip on my network, usually only the one with an SSH connection.

All my WiFi IOTs run on a separate VLAN that right now have a specific pool. If one day I would change the DHCP pool to something else, I would have a lot of OTA to do.

It is nothing wrong with static IPs. But is still another option to be used in the configuration as it would be a OTA_Fallback_IP value. Still, the static IP configuration option already exist.

If you focus your efforts onto the HA add-on as stated some place else, than this should work without a static IP or a working mDNS limitation.

Otherwise thank you for this great project.

miikkajo, 2019-01-16:
i also would like to have this feature, 
for different reason.

1. flashed device got ip from dhcp
2. could not connect with name so i made manual_ip setting for device current ip
3. i want to change ip out of my dhcp reserved scope
4. unable to configure new ip because it's also used in OTA 

Maybe there is a away to change device ip, but i did not figure it out ?

How about configuring ota_ipaddress in yaml config ?
If exists then overrides manually configured ip ?
- miikka